raspiblitz/raspiblitz #3035: Changes to Let's encrypt / duckdns.org settings not possible?

ttrueten, 2022-03-21:
Hi there,
my external IP has changed. However, duckdns.org still points to my old external IP address. Now I thought I would just delete the subscription and set up a new one, but that is not possible because I get the following message here:
```
# running: python /home/admin/config.scripts/blitz.subscriptions.letsencrypt.py subscription-cancel mynode.duckdns.org
  File "/home/admin/config.scripts/blitz.subscriptions.letsencrypt.py", line 70
    print(*args, file=sys.stderr, **kwargs)
          ^
SyntaxError: invalid syntax
# restarting services to pickup new public url (please wait) ...
```
After that the old subscription appears unchanged. What can I do?

Cheers,
Thomas

ttrueten, 2022-03-22:
Hello,
sure. 

Cheers,
Thomas

rootzoll, 2022-04-05:
TODO for 1.8: check automatic IP update when no IP2Tor but duckdns is in use.

maximlomans, 2022-05-02:
Exactly where i just landed two min ago -  just finished installing noip2 - dyndns with whom i have registered a private domain + managed dns service - duckdns can not resolve noip domains | hence the need for a custom dyndns option - possibly via  ~/config.scripts/internet.dyndomain.sh