raspiblitz/raspiblitz #3039: Getting emergency mode screen on boot up every time (CTRL+D to continue works)

kilrau, 2022-03-23:
I am facing https://github.com/rootzoll/raspiblitz/issues/782 / https://github.com/rootzoll/raspiblitz/issues/1053 with a fresh `1.7.2` c-lightning node setup. Not on first boot, but on every boot thereafter. Re-flashed SSD several times using balena etcher & ubuntu disk image writer to no avail.

It's def the sd card that causes it, not the SSD:
[image]

CTRL+D "fixes" it and everything just works fine:
[image]
[image]

Not a perpetual solution though to always have a keyboard on the pi for just this. Manually running `sudo tune2fs -c 1 /dev/mmcblk0p2`, `sudo tune2fs -C 2 -c 1 /dev/mmcblk0p2` and `sudo touch /forcefsck` + reboot didn't help.

Also tried inserting sd card back into computer and running `sudo fsck -f -y /dev/sdb2` (the sd card's `rootfs` partition) which actually seems to have fixed some things:
```
~$ sudo fsck -f -y /dev/sdb2
fsck from util-linux 2.36.1
e2fsck 1.46.3 (27-Jul-2021)
rootfs: recovering journal
Pass 1: Checking inodes, blocks, and sizes
Pass 2: Checking directory structure
Pass 3: Checking directory connectivity
Pass 4: Checking reference counts
Pass 5: Checking group summary information
Free blocks count wrong (60184773, counted=60184769).
Fix? yes

Free inodes count wrong (15436024, counted=15436020).
Fix? yes

rootfs: ***** FILE SYSTEM WAS MODIFIED *****
rootfs: 147436/15583456 files (0.2% non-contiguous), 2270015/62454784 blocks
```

But... still emergency mode on boot.

rootzoll, 2022-03-24:
Did you tried a new/another sd card?

kilrau, 2022-03-24:
Yup, with another sd card it works just fine. Since the underlying issue remains a mystery (the "broken" sd card works flawlessly with a plain debian system on the same pi), should we add an entry in https://github.com/rootzoll/raspiblitz/blob/v1.7/FAQ.md#debug ?

Anyhow, closing here.

closed 2022-03-24 by kilrau

rootzoll, 2022-03-24:
Yeah the reason is hard to say - maybe there is one sector bad on the sd card hardware that is critical when running the raspiblitz image but not when running another image on the same sd card hardware.

If you like to add an entry on this as an PR that would be nice. thx.