raspiblitz/raspiblitz #3040: INFO screen in the LCD display not showing after disabling Lightning capabilities

raulcano, 2022-03-26:
Hi all,

I have a Raspiblitz v1.7.1 running on a Raspberry Pi 4.
The extra services I enabled are JoinMarket and Electrum server.

Last week I disabled Lightning, since I was not making use of it. After that, the screen showing in the LCD is a blue one, instead of the typical INFO screen.

[image]

If I connect to my node via SSH and click on the Info option, I can in fact see the info screen in my laptop, but not in the LCD display:

[image]

I restarted a couple of times and the problem persists.
Also, I tried this:

```
sudo init 3
sudo init 5
restart
```

But no luck either.

Any help will be much appreciated. 
Thanks!

openoms, 2022-04-04:
Not sure if this is fixed in v1.7.2 should test.

rootzoll, 2022-06-05:
I tested again on my machine for - it looks OK so far for 1.8.0. Closing this issue for now - but please reopen if you spot the problem again.

closed 2022-06-05 by rootzoll