raspiblitz/raspiblitz #3041: BTCPayserver invoices broken, stores can't be removed. Possible SQLite related problems.

SomeBTChomer, 2022-03-27:
BTCPayServer v1.4.4 running on Raspiblitz has a few issues for me.

1. Stores can't be removed, from what I could gather, this is related to this implementation using SQLite, which is highly discouraged by the BTCPayserver Team. See this comment from Kukks: https://github.com/btcpayserver/btcpayserver/issues/1424#issuecomment-609559569

2. Lightning invoices created via the BTCPayServer or anything that uses this db are more often than not broken and cannot be decoded by peers. Talking about this with another contributor to BTCPayServer, he pointed out that these issue might also stem from the SQLite implementation.

Here's an example of such a broken invoice: This one was created with RTL.

lnbc110u1p3yqw2rpp5vskrahe73m2c3cpp3tn32yvp38em8upmd38wzyu7t4klu5snvcvqdq8w3jhxaqcqzpgxqrrsssp5qzfnhqf80s87680q3hr5jk35z5nmy2yh2853slwmsskjkg6nhhss9qyyssqwf7a4fxk0en8fxmg4uu0mhh0yjtvyvhh8mx6ta8yxmgt573ah3xjmxdr9fdvu062t6lutk2taa3xdj5nj3dm94ct8nh54ne8zy8we7gqy3sps5

Invoices created by Zeus Wallet, which is also connected to my node work fine, which further confirms this has something to do with the SQLite db, that Zeus probably isn't using to generate its invoices.

Cheers.
Patrick

openoms, 2022-03-30:
Check if LND is running ok:
`lncli getinfo`
Try to reset you macaroons in the CONNECT menu and pair BTCPay with the new connection code.

openoms, 2022-04-04:
You might want reset your BTCPayserver install completely.
To delete the database after you switched it off in SERVICES can remove the data with the command:
```
# WARNING this deletes your BTCPay data
sudo rm  -rf /mnt/hdd/app-data/.btcpayserver
```
and can reinstall afterwards

SomeBTChomer, 2022-04-25:
Sorry for the late reply. I reinstalled the BTCPayserver completely and did set up the store again. Invoices are working most of the time. Not sure why RTL still produces invalid invoices sometimes, but my main issue is fixed. Many thanks for the help.

closed 2022-04-25 by SomeBTChomer