raspiblitz/raspiblitz #3047: High CPU load due to LND on v1.7.1

rene78, 2022-04-01:
I'm experiencing rather high CPU usage. Any idea what can be done to fix it?

[image]
[image]

rene78, 2022-04-01:
Looks like I was able to fix it by changing a setting in the lnd config file:

- Enter `sudo nano .lnd/lnd.conf`
- Add the following line to the config file: `db.bolt.auto-compact=1`

After some compacting during which the CPU load was still high, everything turned back to normal again.

Found here: https://github.com/lightningnetwork/lnd/issues/6065#issuecomment-987699012

rootzoll, 2022-04-02:
Thanks for the report - adressing the lnd database compaction was also part of the v1.7.2 release. Closing issue.

closed 2022-04-02 by rootzoll