raspiblitz/raspiblitz #3048: LetsEncrypt not working

SkabbMask, 2022-04-02:
Hi!
I'm having some issues with setting up the ip2tor bridge combined with the letsencrypt over duckdns, I have very limited experience with network stuff in general so it's probably just something I'm doing wrong but I've spent so many hours troubleshooting it now that I realized I should probably ask for help.

I can access the btcpay server via the ip provided by ip2tor bridge, and the tor address without any problem, as well as through the duckdns subdomain. When I visit it with http:// I get to an nginx start page saying that the web server is working properly, but when I visit it with https:// I get to chose between the two available ip2tor bridges and sign up to pay (again? It's already marked as active in raspiblitz).

I followed this when setting up (and got no errors in that process): https://coincharge.io/en/raspiblitz-btcpay-server/

But now the btcpay server info screen looks like this (when letsencrypt is active).
[image]

Or does the "yet" indicate that I simply just have to wait?

rootzoll, 2022-04-05:
> When I visit it with http:// I get to an nginx start page saying that the web server is working properly, but when I visit it with https:// I get to chose between the two available ip2tor bridges and sign up to pay (again? It's already marked as active in raspiblitz).

Not sure what this means .. how are those options "to chose between" are presented to you from the web browser? Can you share more details?

SkabbMask, 2022-04-06:
I get two buttons to click on, to chose between two different bridges (I assume, that's what it looks like to me - see first screenshot).
If I click on one, I get (second screenshot)
[image]
[image]

rootzoll, 2022-04-25:
Ah OK your operating diretly on the store - I would not recommend that if you want to use IP2Tor with RaspiBlitz. Only use the SSH menu dialogs to setup IP2Tor bridge first and then add duckdns & letsencrpt. Try to follow this tutorial: https://github.com/rootzoll/raspiblitz#ip2tor-paid

SkabbMask, 2022-04-25:
I'm not sure what "operating directly on the store" means. I was following this tutorial: https://coincharge.io/en/raspiblitz-btcpay-server/
And from what I can see in the tutorial (ip2tor bridge part at least), it's the same.
To clarify; I did setup the ip2tor bridge through the raspiblitz. I know no other way of doing it 😅

rootzoll, 2022-04-26:
How do you get the website with the two different bridges to (from the screenshot you posted above). Thats when you use your duckdns domain with https (that you set for IP2Tor)? Do you add the port to the domain? It should be in the end something like this: https://yoursubdomain.duckdns.org:45452

SkabbMask, 2022-04-27:
Yes. Hmm, I don't remember specifying a port so that might be it then, I'll try and get back. Thanks! :)

SkabbMask, 2022-04-27:
[image]

Well, now I can't even get that far >.<
I've probably fucked it up somehow :(

SkabbMask, 2022-05-04:
Update: I get to the sign-in page when I add the port to the URL. But it still says that it's unsecure: https://mhm1234567.duckdns.org:47731/

SkabbMask, 2022-05-04:
It says the cert is issued to "coinpages.duckdns.org"