raspiblitz/raspiblitz #3053: [Research] Use Packer to automate image creation from one config file

openoms, 2022-04-04:
https://www.packer.io/ as recommended by @bodymindarts looks to be hugely useful tool built for just what we need to make image creation easier.

instead of a long and trusted manual process as in https://github.com/rootzoll/raspiblitz/blob/v1.7/FAQ.md#what-is-the-process-of-creating-a-new-sd-card-image-release and  
https://github.com/rootzoll/raspiblitz/pull/3052
the build could be more automated down to a single line from the start including the installation of the base image.

Related issues: 
https://github.com/rootzoll/raspiblitz/issues/2842
https://github.com/rootzoll/raspiblitz/issues/2389
PR: 
https://github.com/rootzoll/raspiblitz/pull/3052

openoms, 2022-04-04:
There has been already an initial config done for packer by @nolith and @geco9, but not sure if it has been used so far: https://github.com/rootzoll/raspiblitz/tree/dev/alternative.platforms/amd64/packer

openoms, 2022-07-30:
there is excellent progress in: https://github.com/rootzoll/raspiblitz/issues/2842

openoms, 2022-11-24:
Managed to set up a workflow on github (which also works locally on my Debian laptop much faster) to generate a qcow2 format amd64 image.
Works when mounted with libvirt / virsh / virt-manager (https://virt-manager.org/), but couldn't figure out how to make a bootable image from it for bare metal. 

tried to convert to a raw image with:
```
qemu-img convert debian-11.5-amd64.qcow2 debian-11.5-amd64.img
```
but can only mount the boot partition from that: 
[image]

Same when writing it to a disk connected with USB straight with:
```
sudo qemu-img dd bs=4M if=debian-11.5-amd64.qcow2  of=/dev/sdd
```
the resulting disk is not bootable for the same reason:
[image]

Would appreciate any help on this how to make a bootable SSD from the .qcow2 file.

## Images generated in github actions
Find the images in the green runs in github actions at:
https://github.com/openoms/raspiblitz/actions

The images are built using rootzoll dev branch, only the workflows are in mine. 
The arm64 workflow is not functional yet.

## Local build
or build your own locally with the [Makefile](https://github.com/openoms/raspiblitz/blob/ci-amd64/Makefile) (needs ~10 GB free space):
```
git clone https://github.com/openoms/raspiblitz
cd raspiblitz
git checkout ci-amd64
make amd-64-image
```

rootzoll, 2022-11-26:
Changing the sd card build process is quite a big refactor .. @openoms do you really want to do this for 1.9.0 .. I would at least recommend to push this to 1.10.0

openoms, 2022-12-08:
The amd64 image creation is now working locally and github actions.

The arm64-rpi image build does run well in GitHub actions now, but not locally yet as don't know how to take this step:
```
      - name: Set up QEMU
        uses: docker/setup-qemu-action@v2
        with:
          platforms: linux/amd64,linux/arm64
```
Related issue to track or contribute to with a solution: https://github.com/mkaczanowski/packer-builder-arm/pull/246

rootzoll, 2022-12-11:
- [ ] check how much the paker script can be used to shorten/optimise the release build process of the sd card

rootzoll, 2022-12-21:
@openoms I tested this automatic lean-build for rpi: https://github.com/rootzoll/raspiblitz/actions/runs/3741413796

I was not able to flash to a 32GB sd card - only a 64GB sd card worked. It looks like the image is of a size of 32.2GB
[image]

When then starting up on a rpi was not able to connect per SSH: `Connection refused`

openoms, 2022-12-21:
Let's reduce it to 28GB then. Can try to debug the image starting as an arm64 VM as well.

rootzoll, 2023-11-14:
@openoms this is now done?

openoms, 2023-11-15:
Yes, I am the images for a while so can say the process is mature enough.

closed 2023-11-15 by openoms