raspiblitz/raspiblitz #3054: CLN v0.11.0 update

openoms, 2022-04-04:
The v0.11.0rc1 is about to be tagged so good to start testing: https://github.com/ElementsProject/lightning/pull/5155

openoms, 2022-04-10:
v0.11.0rc2 installed ok on running RPi and and64 as well with:
```
config.scripts/cl.install.sh update v0.11.0rc2
```

The install script will need to be refactored due to poetry being used to install the python dependencies now: https://github.com/ElementsProject/lightning/blob/master/doc/INSTALL.md#to-build-on-ubuntu

openoms, 2022-05-02:
the update to the final v0.11.0.1 also works well.

- [  ] Will also need to expose the new GRPC interface and
- [  ] Expose Runes (likely a future method to connect Zeus too)

rootzoll, 2022-05-02:
@fusion44 is GRPC/Runes also something to use with the API/library?

fusion44, 2022-05-02:
GRPC definitely. If I knew the GRPC API would come so soon, I'd have waited with the cln implementation. Saves a lot of time compared to the current rpc API.

openoms, 2022-05-05:
Note the rebrand of C-lightning to Core Lightning  (CLN) https://blog.blockstream.com/en-c-lightning-is-now-core-lightning/  also for the webgui @fusion44 
- [ ] will change the naming in the ssh menu

rootzoll, 2022-05-07:
New CLN install does not run thru on sd card build (based on new RaspiOS).  @openoms can you see why `mrkd` is missing here and how to make it available?

```
- Install to /usr/local/bin/
mkdir -p /usr/local/bin
mkdir -p /usr/local/libexec/c-lightning
mkdir -p /usr/local/libexec/c-lightning/plugins
mkdir -p /usr/local/share/man/man1
mkdir -p /usr/local/share/man/man5
mkdir -p /usr/local/share/man/man7
mkdir -p /usr/local/share/man/man8
mkdir -p /usr/local/share/doc/c-lightning
cp tools/hsmtool tools/lightning-hsmtool
install cli/lightning-cli lightningd/lightningd tools/lightning-hsmtool /usr/local/bin
install lightningd/lightning_channeld lightningd/lightning_closingd lightningd/lightning_connectd lightningd/lightning_dualopend lightningd/lightning_gossipd lightningd/lightning_hsmd lightningd/lightning_onchaind lightningd/lightning_openingd lightningd/lightning_websocketd /usr/local/libexec/c-lightning
[ -z "plugins/autoclean plugins/bcli plugins/fetchinvoice plugins/funder plugins/topology plugins/keysend plugins/offers plugins/pay plugins/txprepare plugins/spenderp" ] || install plugins/autoclean plugins/bcli plugins/fetchinvoice plugins/funder plugins/topology plugins/keysend plugins/offers plugins/pay plugins/txprepare plugins/spenderp /usr/local/libexec/c-lightning/plugins
mrkd doc/lightning-cli.1.md
/bin/sh: 1: mrkd: not found
make: *** [doc/Makefile:120: doc/lightning-cli.1] Error 127
```

openoms, 2022-05-07:
> New CLN install does not run thru on sd card build (based on new RaspiOS). @openoms can you see why `mrkd` is missing here and how to make it available?

Can you try:
```
sudo pip3 install --user mrkd==0.2.0
sudo pip3 install --user mistune==0.8.4
```
instead of:
```
sudo -u bitcoin pip3 install --user mrkd==0.2.0
sudo -u bitcoin pip3 install --user mistune==0.8.4
```
(there were both run before #3086)

openoms, 2022-05-07:
The `rustfmt` requirement is a more problematic one as it is not available in the Debian Bullseye package repo only in the unstable repo.
Can install with the following but  there are steps breaking the script and this needs more testing:
```
  # rustfmt from sid(unstable) repo in Debian
  echo "\
Explanation: Uninstall or do not install any Debian-originated
Explanation: package versions other than those in the stable distro
Package: *
Pin: release a=stable
Pin-Priority: 900

Package: *
Pin: release o=Debian
Pin-Priority: -10
" | sudo tee /etc/apt/preferences
  if ! grep -Eq '^deb http://deb.debian.org/debian/ unstable main' /etc/apt/sources.list; then
    echo "deb http://deb.debian.org/debian/ unstable main" | /etc/apt/sources.list
  fi
  sudo apt update
  sudo apt-get -t unstable install -y rustfmt
  ```

rootzoll, 2022-05-07:
I added the proposed changes to the cln install script in the `lightningupdates` branch .. see:
https://github.com/rootzoll/raspiblitz/blob/lightningupdates/home.admin/config.scripts/cl.install.sh

Now on the rustfmt install I get an interactive dialog like this:
[image]

and also at the end ist fails with this:

```
mrkd doc/lightning-cli.1.md
/bin/sh: 1: mrkd: not found
make: *** [doc/Makefile:120: doc/lightning-cli.1] Error 127
rm external/aarch64-linux-gnu/libwally-core-build/src/secp256k1/libsecp256k1.la

- Install to /usr/local/bin/
mkdir -p /usr/local/bin
mkdir -p /usr/local/libexec/c-lightning
mkdir -p /usr/local/libexec/c-lightning/plugins
mkdir -p /usr/local/share/man/man1
mkdir -p /usr/local/share/man/man5
mkdir -p /usr/local/share/man/man7
mkdir -p /usr/local/share/man/man8
mkdir -p /usr/local/share/doc/c-lightning
cp tools/hsmtool tools/lightning-hsmtool
cargo build --quiet --bin cln-grpc
error: failed to parse manifest at `/home/bitcoin/lightning/cln-rpc/Cargo.toml`

Caused by:
  failed to parse the `edition` key

Caused by:
  this version of Cargo is older than the `2021` edition, and only supports `2015` and `2018` editions.
make: *** [plugins/Makefile:194: target/debug/cln-grpc] Error 101
```

openoms, 2022-05-09:
good news, `rustfmt` is part of the Rust install when using the rustup script
mrkd and mistune needs to be globally available for the build
should be fixed in [b2669ed](https://github.com/rootzoll/raspiblitz/commit/b2669eda0d75bbbec0a93baddf78abb66402e733)

rootzoll, 2022-05-09:
great - install works now :) - see mergs in branch `lightningupdates`: https://github.com/rootzoll/raspiblitz/blob/lightningupdates/home.admin/config.scripts/cl.install.sh

rootzoll, 2022-06-05:
With the update to be tested during RC2

openoms, 2022-06-22:
New release soon, tested ok: https://github.com/ElementsProject/lightning/pull/5327#issuecomment-1162431500

openoms, 2022-06-25:
Running
```
config.scripts/cl.install.sh update v0.11.2rc3
```
ok
https://twitter.com/openoms/status/1540573397725462528

openoms, 2022-06-27:
https://github.com/ElementsProject/lightning/releases/tag/v0.11.2

rootzoll, 2022-07-27:
Tested over the last weeks - looks good .. closing for v1.8.0 release.

closed 2022-07-27 by rootzoll