raspiblitz/raspiblitz #3055: BTCPay - NBXPlorer update needs database pinned or migrated to postgesql

openoms, 2022-04-04:
For now to keep the old database format it seems that need to add to the 
`Execstart=` line in the nbxplorer systemd service.
```
--nomigrateevts --nomigraterawtxs
```

https://github.com/dgarage/NBXplorer/blob/master/docs/Postgres-Migration.md

rootzoll, 2022-04-05:
When was this change introduced? Does this effect already people running the v1.7.2?

Bucherbp, 2022-04-07:
This occurred for me when updating BTCPay from 1.7.2

openoms, 2022-04-07:
> When was this change introduced? Does this effect already people running the v1.7.2?

no, it is in the last few versions of NBXplorer, not in the one shipped with v1.7.2.

@Bucherbp for now you can reinstall BTCPay from SERVICES to reset to the pinned version.

Bucherbp, 2022-04-07:
> @Bucherbp for now you can reinstall BTCPay from SERVICES to reset to the pinned version.

Yep, already done, thanks.