raspiblitz/raspiblitz #3058: Raspiblitz - lightning wont finish - going backwards

domodomodomo3, 2022-04-05:
Hej node lovers

my raspiblitz node has a problem. the lightning progress doesnt finish. i tried a restart. i tried updating to the newest version (1.7.2) and tried "bitcoin-cli" because i read it somewhere that it might help. The lightning progress is slowly going backwards. so probably the lightning is not able to catch up with the blockchain. why? this is not my first setup. my node was running perfectly for three years now. the longest i waited for lightning to finish is three weeks now. what can i do? how to fix it? 
here is my log: https://l.termbin.com/8g36

can someone help? 
[image]

rootzoll, 2022-04-25:
Looking at your debug logs .. there are is no systemd report or logs on LND:
```
*** LND (MAINNET) SYSTEMD STATUS ***

*** LAST LND (MAINNET) ERROR LOGS ***
sudo journalctl -u lnd -b --no-pager -n12
-- Journal begins at Mon 2022-04-04 11:34:09 BST, ends at Tue 2022-04-05 15:38:18 BST. --
-- No entries --
```

This is very strange - especially when your getting a percentage displayed. Can you please confirm those debug messages by exiting to terminal (CTRL+c) and then run the command `sudo systemctl status lnd` and paste the output here?

domodomodomo3, 2022-04-25:
Thanks for responding. I still have the same problem. Only change. The lightning progress changed from showing some percentage (which was slowly going backwards) to simply "waiting". It is like this for another week or so. After running the command "sudo systemctl status lnd" i get "Unit lnd.service could not be found".

What can i do to get lightning running again? channel is still funded.

rootzoll, 2022-04-25:
So somehow the install of the LND seemed to got wrong on your update - if even the LND service does not exist. If you have a funded channels first thing would be to make a LND rescue backup - just in case and to check if your LND data on the HDD is still OK. So exit to terminal again with CTRL+c and then just quicly check if you get a listing of the LND directory and wallet data is still accessable with `sudo ls -la /mnt/hdd/lnd` .. let me know if you see files listed there.

If you see files listed type command `menu` > REPAIR > REPAIR-LND > BACKUP-LND and folow the instructions to download the LND rescue file. If everything else fails you have at least your LND wallet secured.

Then I would try a new flashed sd card with the v1.7.2 image (best to use a brand new sd card just in case this error comes from a sd card hardware failure). If you end up with the same result please write back.

domodomodomo3, 2022-04-26:
update: lightning progress is running backward again. now at 98.83%

domodomodomo3, 2022-04-26:
> Looking at your debug logs .. there are is no systemd report or logs on LND:
>
> ```
> *** LND (MAINNET) SYSTEMD STATUS ***
>
> *** LAST LND (MAINNET) ERROR LOGS ***
> sudo journalctl -u lnd -b --no-pager [...]

rootzoll, 2022-04-26:
Can you try `sudo systemctl status lnd` again - is the lnd systemd still there?

rootzoll, 2022-04-26:
If lnd is running as systemd - can you know check and post the debug info again?

domodomodomo3, 2022-04-27:
thx for your help. here it is. https://l.termbin.com/g9n0

domodomodomo3, 2022-05-13:
can anyone help?

openoms, 2022-05-17:
look at the lnd log further in the SYSTEM menu. See what errors you as there was nothing caught in your debug log.
 
 Might be beneficial to reduce the lightning logs to info from debug in the lnd.conf as it migth  just be too verbose.

domodomodomo3, 2022-05-17:
**### **here is what i get from: sudo systemctl status lnd****
 
● lnd.service - Lightning Network Daemon
     Loaded: loaded (/etc/systemd/system/lnd.service; enabled; vendor preset: e>
     Active: active (running) since Tue 2022-05-10 11:21:13 BST; 1 weeks 0 days>
    Process: 3648856 ExecStartPre=/home/admin/config.scripts/lnd.check.sh prest>
   Main PID: 3649061 (lnd)
     Status: "Wallet unlocked"
      Tasks: 11 (limit: 4532)
        CPU: 1h 22min 50.151s
     CGroup: /system.slice/lnd.service
             └─3649061 /usr/local/bin/lnd --configfile=/home/bitcoin/.lnd/lnd.c>

Warning: journal has been rotated since unit was started, output may be incompl>
lines 1-12/12 (END)

**### and here is what i get from the SYSTEM menu lnd log:**

2022-05-17 18:41:59.145 [DBG] RPCS: [channelbalance] local_balance=246375000 mSAT remote_balance=0 mSAT unsettled_local_balance=0 mSAT unsettled_remote_balance=0 mSAT pending_open_local_balance=0 mSAT pending_open_remote_balance=0 mSAT
2022-05-17 18:42:30.214 [DBG] RPCS: [walletbalance] Total balance=0.00001166 BTC (confirmed=0.00001166 BTC, unconfirmed=0 BTC)
2022-05-17 18:42:30.571 [DBG] RPCS: [channelbalance] local_balance=246375000 mSAT remote_balance=0 mSAT unsettled_local_balance=0 mSAT unsettled_remote_balance=0 mSAT pending_open_local_balance=0 mSAT pending_open_remote_balance=0 mSAT
2022-05-17 18:42:57.722 [DBG] RPCS: [feereport]
2022-05-17 18:43:06.448 [DBG] RPCS: [walletbalance] Total balance=0.00001166 BTC (confirmed=0.00001166 BTC, unconfirmed=0 BTC)
2022-05-17 18:43:06.836 [DBG] RPCS: [channelbalance] local_balance=246375000 mSAT remote_balance=0 mSAT unsettled_local_balance=0 mSAT unsettled_remote_balance=0 mSAT pending_open_local_balance=0 mSAT pending_open_remote_balance=0 mSAT
2022-05-17 18:43:38.403 [DBG] RPCS: [walletbalance] Total balance=0.00001166 BTC (confirmed=0.00001166 BTC, unconfirmed=0 BTC)
2022-05-17 18:43:38.770 [DBG] RPCS: [channelbalance] local_balance=246375000 mSAT remote_balance=0 mSAT unsettled_local_balance=0 mSAT unsettled_remote_balance=0 mSAT pending_open_local_balance=0 mSAT pending_open_remote_balance=0 mSAT
2022-05-17 18:44:13.683 [DBG] RPCS: [walletbalance] Total balance=0.00001166 BTC (confirmed=0.00001166 BTC, unconfirmed=0 BTC)
2022-05-17 18:44:14.090 [DBG] RPCS: [channelbalance] local_balance=246375000 mSAT remote_balance=0 mSAT unsettled_local_balance=0 mSAT unsettled_remote_balance=0 mSAT pending_open_local_balance=0 mSAT pending_open_remote_balance=0 mSAT
2022-05-17 18:44:38.406 [DBG] RPCS: [feereport]
2022-05-17 18:44:49.886 [DBG] RPCS: [walletbalance] Total balance=0.00001166 BTC (confirmed=0.00001166 BTC, unconfirmed=0 BTC)
2022-05-17 18:44:50.243 [DBG] RPCS: [channelbalance] local_balance=246375000 mSAT remote_balance=0 mSAT unsettled_local_balance=0 mSAT unsettled_remote_balance=0 mSAT pending_open_local_balance=0 mSAT pending_open_remote_balance=0 mSAT
2022-05-17 18:45:22.834 [DBG] RPCS: [walletbalance] Total balance=0.00001166 BTC (confirmed=0.00001166 BTC, unconfirmed=0 BTC)
2022-05-17 18:45:23.240 [DBG] RPCS: [channelbalance] local_balance=246375000 mSAT remote_balance=0 mSAT unsettled_local_balance=0 mSAT unsettled_remote_balance=0 mSAT pending_open_local_balance=0 mSAT pending_open_remote_balance=0 mSAT
2022-05-17 18:45:57.037 [DBG] RPCS: [walletbalance] Total balance=0.00001166 BTC (confirmed=0.00001166 BTC, unconfirmed=0 BTC)
2022-05-17 18:45:57.471 [DBG] RPCS: [channelbalance] local_balance=246375000 mSAT remote_balance=0 mSAT unsettled_local_balance=0 mSAT unsettled_remote_balance=0 mSAT pending_open_local_balance=0 mSAT pending_open_remote_balance=0 mSAT
2022-05-17 18:46:18.228 [DBG] RPCS: [feereport]
2022-05-17 18:46:28.402 [DBG] RPCS: [walletbalance] Total balance=0.00001166 BTC (confirmed=0.00001166 BTC, unconfirmed=0 BTC)
2022-05-17 18:46:28.763 [DBG] RPCS: [channelbalance] local_balance=246375000 mSAT remote_balance=0 mSAT unsettled_local_balance=0 mSAT unsettled_remote_balance=0 mSAT pending_open_local_balance=0 mSAT pending_open_remote_balance=0 mSAT
2022-05-17 18:47:02.205 [DBG] RPCS: [walletbalance] Total balance=0.00001166 BTC (confirmed=0.00001166 BTC, unconfirmed=0 BTC)
2022-05-17 18:47:02.626 [DBG] RPCS: [channelbalance] local_balance=246375000 mSAT remote_balance=0 mSAT unsettled_local_balance=0 mSAT unsettled_remote_balance=0 mSAT pending_open_local_balance=0 mSAT pending_open_remote_balance=0 mSAT
2022-05-17 18:47:34.313 [DBG] RPCS: [walletbalance] Total balance=0.00001166 BTC (confirmed=0.00001166 BTC, unconfirmed=0 BTC)
2022-05-17 18:47:34.677 [DBG] RPCS: [channelbalance] local_balance=246375000 mSAT remote_balance=0 mSAT unsettled_local_balance=0 mSAT unsettled_remote_balance=0 mSAT pending_open_local_balance=0 mSAT pending_open_remote_balance=0 mSAT
2022-05-17 18:47:58.888 [DBG] RPCS: [listchannels] fetched 1 channels from DB
2022-05-17 18:47:59.896 [DBG] RPCS: [feereport]
2022-05-17 18:48:12.948 [DBG] RPCS: [walletbalance] Total balance=0.00001166 BTC (confirmed=0.00001166 BTC, unconfirmed=0 BTC)
2022-05-17 18:48:13.306 [DBG] RPCS: [channelbalance] local_balance=246375000 mSAT remote_balance=0 mSAT unsettled_local_balance=0 mSAT unsettled_remote_balance=0 mSAT pending_open_local_balance=0 mSAT pending_open_remote_balance=0 mSAT
2022-05-17 18:48:45.801 [DBG] RPCS: [walletbalance] Total balance=0.00001166 BTC (confirmed=0.00001166 BTC, unconfirmed=0 BTC)
[... cut ...]
2022-05-17 18:49:57.411 [DBG] RPCS: [channelbalance] local_balance=246375000 mSAT remote_balance=0 mSAT unsettled_local_balance=0 mSAT unsettled_remote_balance=0 mSAT pending_open_local_balance=0 mSAT pending_open_remote_balance=0 mSAT
2022-05-17 18:50:32.827 [DBG] RPCS: [walletbalance] Total balance=0.00001166 BTC (confirmed=0.00001166 BTC, unconfirmed=0 BTC)
2022-05-17 18:50:33.241 [DBG] RPCS: [channelbalance] local_balance=246375000 mSAT remote_balance=0 mSAT unsettled_local_balance=0 mSAT unsettled_remote_balance=0 mSAT pending_open_local_balance=0 mSAT pending_open_remote_balance=0 mSAT

domodomodomo3, 2022-05-17:
**### And here is the lnd.conf**

Editing the /mnt/hdd/lnd/lnd.conf

│ │# lnd configuration                                                                                   │ │
│ │                                                                                                      │ │
│ │[Application Options]                                                                                 │ │
│ │accept-keysend=true                                                                                   │ │
│ │listen=0.0.0.0:9735                                                                                   │ │
│ │tlsdisableautofill=1                                                                                  │ │
│ │tlsautorefresh=1                                                                                      │ │
│ │stagger-initial-reconnect=1                                                                           │ │
│ │sync-freelist=1                                                                                       │ │
│ │ignore-historical-gossip-filters=1                                                                    │ │
│ │debuglevel=debug                                                                                      │ │
│ │maxpendingchannels=5                                                                                  │ │
│ │alias=blackbox                                                                                        │ │
│ │color=#68F442                                                                                         │ │
│ │nat=false                                                                                             │ │
│ │                                                                                                      │ │
│ │# RPC open to all connections on Port 10009                                                           │ │
│ │rpclisten=0.0.0.0:10009                                                                               │ │
│ │# REST open to all connections on Port 8080                                                           │ │
│ │restlisten=0.0.0.0:8080                                                                               │ │
│ │# Domain, could use https://freedns.afraid.org                                                        │ │
│ │#tlsextradomain=lightning.yourhost.com                                                                │ │
│ │tlsextraip=0.0.0.0                                                                                    │ │
│ │                                                                                                      │ │
│ │[Bitcoin]                                                                                             │ │
│ │bitcoin.active=1                                                                                      │ │
│ │bitcoin.node=bitcoind                                                                                 │ │
│ │# enable either testnet or mainnet                                                                    │ │
│ │#bitcoin.testnet=1                                                                                    │ │
│ │bitcoin.mainnet=1                                                                                     │ │
│ │                                                                                                      │ │
│ │                                                                                                      │ │
│ │[autopilot]                                                                                           │ │
│ │autopilot.active=1                                                                                    │ │
│ │autopilot.maxchannels=5                                                                               │ │
│ │autopilot.allocation=0.6                                                                              │ │
│ │                                                                                                      │ │
│ │[bitcoind]                                                                                            │ │
│ │bitcoind.rpchost=127.0.0.1:8332                                                                       │ │
│ │bitcoind.rpcuser=raspibolt                                                                            │ │
│ │bitcoind.rpcpass=btcnodebtc                                                                           │ │
│ │bitcoind.zmqpubrawblock=tcp://127.0.0.1:28332                                                         │ │
│ │bitcoind.zmqpubrawtx=tcp://127.0.0.1:28333                                                            │ │
│ │                                                                                                      │ │
│ │                                                                                                      │ │
│ │[bolt]                                                                                                │ │
│ │db.bolt.auto-compact=true                                                                             │ │
│ │db.bolt.auto-compact-min-age=672h                                                                     │ │
│ │                                                                                                      │ │
[... cut ...]
│ │tor.active=true                                                                                       │ │
│ │tor.v3=true                                                                                           │ │
│ │tor.privatekeypath=/mnt/hdd/lnd/v3_onion_private_key                                                  │ │
│ │tor.socks=9050                                                                                        │ │
│ │tor.control=9051                                                                                      │ │
│ │

openoms, 2022-05-17:
now edit in the lnd.conf
`debuglevel=info`

restart lnd
sudo systemctl restart lnd

watch the logs in one window

and can unlock in an other.

Would also advise against using the LND autopilot.

domodomodomo3, 2022-05-18:
**edited to debuglevel=info
restarted lnd.
still looks likes lightning cant follow up with the blockchain.

here are the logs:**

2022-05-18 12:25:52.356 [INF] CHDB: Not compacting database file at /home/bitcoin/.lnd/data/graph/mainnet/channel.db, it was last compacted at 2022-05-10 11:21:10.755689989 +0100 BST (193h4m41s ago), min age is set to 672h0m0s
2022-05-18 12:25:52.356 [INF] CHDB: Not compacting database file at /home/bitcoin/.lnd/data/chain/bitcoin/mainnet/macaroons.db, it was last compacted at 2022-05-10 11:21:10.797706846 +0100 BST (193h4m41s ago), min age is set to 672h0m0s
2022-05-18 12:25:52.356 [INF] CHDB: Not compacting database file at /home/bitcoin/.lnd/data/graph/mainnet/sphinxreplay.db, it was last compacted at 2022-05-10 11:21:10.823924068 +0100 BST (193h4m41s ago), min age is set to 672h0m0s
2022-05-18 12:25:52.357 [INF] LTND: Creating local graph and channel state DB instances
2022-05-18 12:25:54.263 [INF] CHDB: Checking for schema update: latest_version=24, db_version=24
2022-05-18 12:25:54.263 [INF] LTND: Database(s) now open (time_to_open=1.907101849s)!
2022-05-18 12:25:54.263 [INF] LTND: Systemd was notified about our readiness
2022-05-18 12:25:54.263 [INF] LTND: Waiting for wallet encryption password. Use `lncli create` to create a wallet, `lncli unlock` to unlock an existing wallet, or `lncli changepassword` to change the password of an existing wallet and unlock it.
2022-05-18 12:25:57.815 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: wallet locked, unlock it to enable full RPC access
2022-05-18 12:26:05.064 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: wallet locked, unlock it to enable full RPC access
2022-05-18 12:26:05.303 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: wallet locked, unlock it to enable full RPC access
2022-05-18 12:26:06.473 [ERR] RPCS: [/lnrpc.Lightning/GetInfo]: wallet locked, unlock it to enable full RPC access
2022-05-18 12:26:08.418 [INF] LNWL: Opened wallet
2022-05-18 12:26:08.661 [INF] CHRE: Primary chain is set to: bitcoin
2022-05-18 12:26:08.693 [INF] CHRE: Initializing bitcoind backed fee estimator in CONSERVATIVE mode
2022-05-18 12:26:08.693 [INF] LNWL: Started listening for bitcoind block notifications via ZMQ on 127.0.0.1:28332
2022-05-18 12:26:08.696 [INF] LNWL: Started listening for bitcoind transaction notifications via ZMQ on 127.0.0.1:28333
2022-05-18 12:26:11.508 [INF] LNWL: The wallet has been unlocked without a time limit
2022-05-18 12:26:11.517 [INF] CHRE: LightningWallet opened
2022-05-18 12:26:11.524 [INF] SRVR: Proxying all network traffic via Tor (stream_isolation=true)! NOTE: Ensure the backend node is proxying over Tor as well
2022-05-18 12:26:11.525 [INF] TORC: Starting tor controller
2022-05-18 12:26:11.532 [INF] HSWC: Cleaning circuits from disk for closed channels
2022-05-18 12:26:11.537 [INF] HSWC: Finished cleaning: num_closed_channel=2, num_circuits=0, num_keystone=0
2022-05-18 12:26:11.537 [INF] HSWC: Restoring in-memory circuit state from disk
2022-05-18 12:26:11.540 [INF] HSWC: Payment circuits loaded: num_pending=0, num_open=0
2022-05-18 12:26:11.542 [INF] HSWC: Trimming open circuits for chan_id=666741:1199:1, start_htlc_id=0
2022-05-18 12:26:11.561 [INF] LTND: Channel backup proxy channel notifier starting
2022-05-18 12:26:11.561 [INF] ATPL: Instantiating autopilot with active=false, max_channels=5, allocation=0.600000, min_chan_size=20000, max_chan_size=16777215, private=false, min_confs=1, conf_target=3
2022-05-18 12:26:11.562 [INF] LTND: Systemd was notified about our readiness
2022-05-18 12:26:11.566 [INF] LTND: Waiting for chain backend to finish sync, start_height=725342
2022-05-18 12:28:31.549 [INF] LTND: Received terminated
2022-05-18 12:28:31.549 [INF] LTND: Shutting down...
2022-05-18 12:28:31.549 [INF] LTND: Systemd was notified about stopping
2022-05-18 12:28:31.549 [INF] LTND: Gracefully shutting down.
2022-05-18 12:28:32.365 [INF] RPCS: Stopping RPC Server
2022-05-18 12:28:32.365 [INF] RPCS: Stopping WalletKitRPC Sub-RPC Server
2022-05-18 12:28:32.365 [INF] RPCS: Stopping AutopilotRPC Sub-RPC Server
2022-05-18 12:28:32.365 [INF] RPCS: Stopping ChainRPC Sub-RPC Server
2022-05-18 12:28:32.365 [INF] RPCS: Stopping InvoicesRPC Sub-RPC Server
2022-05-18 12:28:32.366 [INF] RPCS: Stopping SignRPC Sub-RPC Server
2022-05-18 12:28:32.366 [INF] RPCS: Stopping RouterRPC Sub-RPC Server
2022-05-18 12:28:32.366 [INF] RPCS: Stopping VersionRPC Sub-RPC Server
2022-05-18 12:28:32.366 [INF] RPCS: Stopping WatchtowerRPC Sub-RPC Server
2022-05-18 12:28:32.366 [INF] RPCS: Stopping WatchtowerClientRPC Sub-RPC Server
2022-05-18 12:28:32.366 [INF] TORC: Stopping tor controller
2022-05-18 12:28:32.367 [ERR] TORC: DEL_ONION got error: invalid arguments: unexpected code
2022-05-18 12:28:32.367 [ERR] LTND: error stopping tor controller: invalid arguments: unexpected code
2022-05-18 12:28:32.386 [INF] LTND: Shutdown complete

2022-05-18 12:28:33.245 [INF] LTND: Version: 0.14.2-beta commit=v0.14.2-beta, build=production, logging=default, debuglevel=info
2022-05-18 12:28:33.246 [INF] LTND: Active chain: Bitcoin (network=mainnet)
2022-05-18 12:28:33.248 [INF] RPCS: RPC server listening on 0.0.0.0:10009
2022-05-18 12:28:33.252 [INF] RPCS: gRPC proxy started at 0.0.0.0:8080
2022-05-18 12:28:33.252 [INF] LTND: Opening the main database, this might take a few minutes...
2022-05-18 12:28:33.252 [INF] LTND: Opening bbolt database, sync_freelist=true, auto_compact=true
[... cut ...]
2022-05-18 12:28:54.183 [INF] HSWC: Trimming open circuits for chan_id=666741:1199:1, start_htlc_id=0
2022-05-18 12:28:54.201 [INF] LTND: Channel backup proxy channel notifier starting
2022-05-18 12:28:54.202 [INF] ATPL: Instantiating autopilot with active=false, max_channels=5, allocation=0.600000, min_chan_size=20000, max_chan_size=16777215, private=false, min_confs=1, conf_target=3
2022-05-18 12:28:54.203 [INF] LTND: Systemd was notified about our readiness
2022-05-18 12:28:54.206 [INF] LTND: Waiting for chain backend to finish sync, start_height=725342

domodomodomo3, 2022-05-18:
debug log from today: https://l.termbin.com/c373

rootzoll, 2022-05-18:
Seeing a `ERROR: AcceptBlockHeader: block 00000000000000000006ef4dd4e882ecd0dd4965a5e1b2bf1d7a2c6a500acaa4 is marked invalid` error in your Bitcoin logs.

Some report that a `reindex` on the blockchain helped to get rid of this issue:
https://github.com/bitcoin/bitcoin/issues/21568#issuecomment-818767509

There is an old raspiblitz script you can try to trigger a reindex `/home/admin/config.scripts/network.reindex.sh` .. but I havent tested that one in a long time.

domodomodomo3, 2022-05-30:
thanks but it didnt worked out. it got stuck in an error loop. did a full reset. now its back online and works.

closed 2022-05-30 by rootzoll