raspiblitz/raspiblitz #3065: UX: Main menu back button is not working properly.

MiguelMedeiros, 2022-04-15:
### Description
When I click on the `Main menu` button on any dialog screen the script stops and exit directly to terminal.

### Steps to reproduce
1- open `./00mainMenu.sh`
2- select `SYSTEM` option
3- select the `<Main menu>` option

https://user-images.githubusercontent.com/5798170/163505455-7ef2c58f-0102-4b1c-8979-727aa7449b11.mp4

### Expected behaviour
The "back to main menu" option should go back to `./00mainMenu.sh` script.

https://user-images.githubusercontent.com/5798170/163505439-c3de5257-e257-4a9a-b21c-810702961457.mp4

### Possible solution
At the end of the file `./99systemMenu.sh` add the following code:

```bash
# go back to main menu (and show)
/home/admin/00mainMenu.sh
exit 0
;;
```
[image]

### Observation
Maybe we should fix all the other `Main menu` and `back menu` buttons.

MiguelMedeiros, 2022-04-15:
I noticed that after syncing everything this bug stopped happening. :thinking:

rootzoll, 2022-04-26:
Normally `00mainMenu.sh` gets not called during sync and is just ready after sync. You can call it manually also with shortcut `menu`, but its more for special debug/repair operations. Normally it will just be called internally when you call `raspiblitz` in a loop and then the back button returns to main menu.

closed 2022-04-26 by rootzoll