raspiblitz/raspiblitz #3066: Copy Blockchain: IP address not shown in rsync command

cstenglein, 2022-04-16:
When copying the blockchain to a newly setup blitz, the instructions just show `bitcoin@:/mnt/hdd/blockchain` without the current IP address after the `@`.

As a user I would like to just copy & paste the command without adding the IP: As an alternative, there could be a placeholder instead of an empty field.

Just the rsync command:
`sudo rsync -avhW --progress ./chainstate ./blocks bitcoin@:/mnt/hdd/bitcoin`

Full text on Linux:
```
****************************************************************************
Instructions to COPY/TRANSFER SYNCED BLOCKCHAIN from a LINUX computer
****************************************************************************

ON YOUR LINUX COMPUTER download and validate the blockchain with the Bitcoin
Core wallet software (>=0.17.1) from: bitcoincore.org/en/download
If the Bitcoin Blockchain is synced up - make sure that your Linux computer &
your RaspiBlitz are in the same local network.

Open a fresh terminal on your Linux computer and change into the directory that
contains the blockchain data - should see folders named 'blocks' & 'chainstate'
there. Normally on Linux thats: cd ~/.bitcoin/
Make sure that the Bitcoin Core Wallet is not running in the background anymore.

COPY, PASTE & EXECUTE the following command on your Linux terminal:
sudo rsync -avhW --progress ./chainstate ./blocks bitcoin@:/mnt/hdd/bitcoin

You will be asked for passwords. First can be the user password of your Linux
computer and the last is the PASSWORD A (or 'raspiblitz') of this RaspiBlitz.

It can take multiple hours until transfer is complete - be patient.
****************************************************************************
PRESS ENTER if transfers is done OR if you want to choose another option.
```

rootzoll, 2022-05-17:
done - needs testing

rootzoll, 2022-06-04:
worked - closing for v1.8.0 release

closed 2022-06-04 by rootzoll