raspiblitz/raspiblitz #3069: LND Rest - Over HTTPs with Lets Encrypt

PrinzMonty, 2022-04-22:
Hello,

I have a Raspiblitz running locally on my network without TOR, I tunnel this into Clearnet via an OpenVPN connection. This works so far and I can work with LNbits for example.

But this works in the background only via the local TLS certificate of the Raspiblitz, but for LNUrl to work a TOR or HTTPS (with top-level domain) connection is necessary.

Is there any way to store a Lets Encrypt certificate from the LND Rest API on the Raspiblitz or does that generally not work with my VPN server in between?

My setup looks like:

[image]

PrinzMonty, 2022-04-23:
I had to read a bit about LNurl and now I understand that my problem has nothing to do with my setup, but something to do with my NGINX LNbits configuration.

I could solve it now ;)

closed 2022-04-23 by PrinzMonty

rootzoll, 2022-04-25:
If you need letsencrypt we have a script that tries to support you on this by using DNS validation if your not running LNbits on port 80. Its basically optimized to work with IP2Tor at the moment but could be extended to also work with custom VPN tunneling - just leaving this here if you wanna take a deeper look and like to commit back: https://github.com/rootzoll/raspiblitz/blob/dev/home.admin/config.scripts/bonus.letsencrypt.sh

PrinzMonty, 2022-04-25:
I managed it without Let's Encrypt. The error was in the configuration of my NGINX proxy, I set the wrong header ;-)

Of course, it would be nice if you could create a Let's Encrypt certificate without IP2TOR, because I think that should also be possible via my VPN path. But that doesn't work because Raspiblitz only accepts it with IP2TOR.

But I think then I would have to redirect the traffic somehow via SOCKS5.... Nuja different topic.

If I somehow find time, I will write a tutorial on how I did it.