raspiblitz/raspiblitz #3071: Files not beign persisted on hdd

jaonoctus, 2022-04-23:
After the boot and inital config from scratch, if I restart the machine before bitcoind is sincronized, the `/mnt/hdd/bitcoin` folder and many others are missing.

And even if I try to recreate them, after the restart the data there will be lost.

rootzoll, 2022-04-25:
This could be related to your hardware setup (maybe the HDD7SSD controller) what parts are you using? Are they as recommended on the parts list or is it even ordered as a pre-assembled node? https://github.com/rootzoll/raspiblitz#amazon-shopping-list-buy-parts--build-it-yourself

godidev, 2022-05-02:
Hi! @jaonoctus Did you find a solution? maybe trying different hardware?

I'm having the same problem. Tried to burn raspiblitz img in the sd again and used the same ssd. 
It insta detected the blockcahim and could continue the blockchain syncro, but 5hours later `/mnt/hdd/bitcoin` folder is missing again xd

I'm not using the recommended parts btw, just the parts I already had at home.

jaonoctus, 2022-05-02:
@godidev not yet.

> I'm not using the recommended parts btw, just the parts I already had at home.

Same

cstenglein, 2023-06-07:
@jaonoctus  is the issue still present in 1.9.0?

closing for now, if it still appears, please re-open the issue. :)