raspiblitz/raspiblitz #3072: CLBOSS update

openoms, 2022-04-25:
New release:
https://github.com/ZmnSCPxj/clboss/releases/tag/0.11D

to update manually need a small edit to download the tarball:
```
nano config.scripts/cl-plugin.clboss.sh
```
change to:
```
CLBOSSVERSION="0.11D"
```

and remove a `v` from the download link so it is:
```
    # download tarball
    sudo -u bitcoin wget \
     https://github.com/ZmnSCPxj/clboss/releases/download/${CLBOSSVERSION}/clboss-${CLBOSSVERSION}.tar.gz \
     -O /home/bitcoin/cl-plugins-available/clboss-${CLBOSSVERSION}.tar.gz || exit 1
```

can use the menu to uninstall then install or run:
```
config.scripts/cl-plugin.clboss.sh off
config.scripts/cl-plugin.clboss.sh on
```

Surenic, 2022-04-25:
Attention: 0.11D has a small bug which prevents clboss configs in the CLCONF to load. At least I tested clboss-min-onchain, clboss-zerobasefee and clboss-min-channel.

Fix is on the way and will be implemented in 0.11E soon

jayrockk, 2022-04-25:
> Attention: 0.11D has a small bug which prevents clboss configs in the CLCONF to load. At least I tested clboss-min-onchain, clboss-zerobasefee and clboss-min-channel.
> 
> Fix is on the way and will be implemented in 0.11E soon

Thanks for the heads up.

openoms, 2022-04-27:
https://github.com/ZmnSCPxj/clboss/releases/tag/0.11E is out

jayrockk, 2022-04-27:
Thanks. Installing now.

rootzoll, 2022-05-02:
see PR #3080

openoms, 2022-05-13:
In the meantime CLBOSS v0.13A is out with some economic improvements. Will include in v1.8.0rc1 https://github.com/ZmnSCPxj/clboss/releases/tag/0.13A

openoms, 2022-05-13:
updated CLBOSS to to 0.13A in both `v1.7` 78a0ae00  and `dev` ea88df84 so people running the patched version can use the more economically viable version.

rootzoll, 2022-07-27:
So far now bad reports over release candidates - closing issue for v1.8.0

closed 2022-07-27 by rootzoll