raspiblitz/raspiblitz #3081: Installing Raspiblitz amd64 image on Digital Ocean - hddCandidate is wrong.

motolese, 2022-05-01:
I successfully imported the raspiblitz_amd64 image in Digital Ocean and installed it in a very cheap instance. 
This can be very useful for some use cases. 

When I try to run a FRESH INSTALL, the script selects VDB as hddCandidate, and the correct disk is SDA. 
sda          8:0         0  600G  0 disk 
vda          254:0     0   80G  0 disk
└─vda1   254:1     0 48.8G 0 part 
vdb          254:16   0  452K  1 disk 

Then I got this error. 

```
Running Format: (ext4) (vdb)
# DATA DRIVE - FORMATTING to EXT4 layout (old)
(standard_in) 1: syntax error
# Unmounting all partitions of this device
# Cleaning /etc/fstab from 2022-05-01-13-34-27-00
# Wiping all partitions (sfdisk/wipefs)
wipefs: /dev/vdb: failed to erase iso9660 magic string at offset 0x00008001: Operation not permitted
Error: Operation not permitted during write on /dev/vdb
# Creating the one big partition
Error: /dev/vdb: unrecognised disk label
# waiting until the partition gets available
# waiting until the partition gets available
# waiting until the partition gets available
# waiting until the partition gets available
# waiting until the partition gets available
# waiting until the partition gets available
# waiting until the partition gets available
# waiting until the partition gets available
# waiting until the partition gets available
# waiting until the partition gets available
# waiting until the partition gets available
# partition failed
FAIL ON FORMATTING THE DRIVE:
partition failed
Please report as issue on the raspiblitz github.
```

In order to select the right hddCandidate, on `controlSetupDialog.sh` I commented the line 147 and added `hddCandidate="sda"` on line 148, so the script goes on smoothly, and format the drive SDA, ask for the passwords,   

Here is the code changed
147 `#    source <(/home/admin/_cache.sh get hddGotMigrationData hddBlocksBitcoin hddBlocksLitecoin hddCandidate)`
148 `hddCandidate="sda"`

Then it fails to mount the disk and the script stops on the Provisioning part.  Here is the raspiblitz.log
https://pastebin.com/0MjjwJmr

Which file do I need to change in order to mount the disk correctly, or tell the script to use sda as hddCandidate for the entire process?

openoms, 2022-05-01:
This is slightly complicated - best would be if there would be a manual selection menu to choose the partition, but for now you are best to
create a partition table to have `sda1` and format to EXT4 and mount it to `/mnt/hdd/' before starting the menu again.

openoms, 2022-05-01:
Command roughly:
```
 parted -s /dev/sda mkpart primary ext4 1024KiB 100%
 mkfs.ext4 -F -L BLOCKCHAIN /dev/sda1
 
# to mount permanentrly thsi script edits the `/ect/fstab`
sudo /home/admin/config.scripts/blitz.datadrive.sh fstab sda1
```

motolese, 2022-05-01:
Thanks @openoms 
I did the commands but I still get an error.  Here is a part of the log, when I try to run the menu again

```
# lsblk -o NAME,FSTYPE,LABEL 
NAME   FSTYPE  LABEL
sda            
└─sda1 ext4    BLOCKCHAIN
vda            
└─vda1 ext4    
vdb    iso9660 config-2
# RASPIBLITZ DATA DRIVE Status

# BASICS
isMounted=0
isBTRFS=0

# SETUP INFO
isSSD=0
hddCandidate='vdb'
hddBytes=
hddGigaBytes=
hddPartitionCandidate=''
hddAdapterUSB=''
hddAdapterUSAP=0

# RAID
isRaid=0
raidCandidate[0]='sda 600 GB DO'
raidCandidates=1

# SWAP
isSwapExternal=0

Displaying Info Message: Please Login for Setup
## WAIT LOOP: USER SETUP/UPDATE/MIGRATION
# PROVISION PROCESS with setupPhase(setup)
Temp mounting (2) data drive (hddFormat='')
Temp mounting (2) result: 0
###########################################################
# blitz.datadrive.sh status
# RASPIBLITZ DATA DRIVE Status

# BASICS
isMounted=1
isBTRFS=0
hddRaspiData=0
hddRaspiVersion=''
isSSD=0
datadisk='sda'
datapartition='sda1'
hddCandidate='sda'
hddPartitionCandidate='sda1'
hddBlocksBitcoin=0
hddBytes=644242997248
hddGigaBytes=599
hddUsedInfo='28K (1%)'
hddDataFreeBytes=600753549312
hddDataFreeKB=586673388
hddDataFreeGB=559
hddAdapterUSB=''
hddAdapterUSAP=0

# RAID
isRaid=0
raidCandidate[0]='vdb 0 GB 0x1af4'
raidCandidates=1

# SWAP
isSwapExternal=0

***********************************************************
ERROR - please report to development team
***********************************************************
state(errorHDD) message(Was not able to mount HDD (2))
```

openoms, 2022-05-01:
is the `/dev/sda1` mounted on `/mnt/hdd` ?

see:
`df -h`

motolese, 2022-05-01:
Yes is mounted. This vdb is created by DigitalOcean, and its sucks. 

Now I got the message of an inconsistent system. The instance is refreshed waiting for instructions. 
https://imgur.com/a/gzwfOKb 

Also, I tried to change vdb to sda and sda to vdb following this but wont work too. 
https://snipplr.com/view/80310/replace-devsda-with-devsdb-device-name-assignment

There is possible to set the script to use sda instead vdb?
The problem is here raidCandidate[0]='vdb 0 GB 0x1af4'

motolese, 2022-05-01:
```
admin@165.232.147.84:~₿ df -h
Filesystem      Size  Used Avail Use% Mounted on
udev            1.9G     0  1.9G   0% /dev
tmpfs           394M  1.2M  393M   1% /run
/dev/vda1       7.4G  6.1G  878M  88% /
tmpfs           2.0G   16K  2.0G   1% /dev/shm
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs            32M  8.0K   32M   1% /var/cache/raspiblitz
tmpfs           394M  132K  394M   1% /run/user/1001
/dev/sda1       590G   28K  560G   1% /mnt/hdd
```

openoms, 2022-05-01:
So at which step are you stuck?

the easiest to do is to hardcode some values I think

You can modify the values in
`config.scripts/blitz.datadrive.sh` 

The values are given in the lines like:
```
echo "hddCandidate='${hdd}'"
 
echo "isMounted=${isMounted}"
```
or anything you need

motolese, 2022-05-01:
I created the 3 passwords, and then it fails. 
I will make the changes in this file `config.scripts/blitz.datadrive.sh` and `home.admin/_bootstrap.sh` too, before running.

motolese, 2022-05-01:
No way, several different errors, and after restarting this screen! 
https://imgur.com/a/gzwfOKb 

In the log `provision did not ran thru - need fresh sd card`.

openoms, 2022-05-02:
You can try to reset this by deleting the file:
`sudo rm /home/admin/provision.flag`

Other files used to record the state (so can be deleted to restart the setup process) are:
```
/home/admin/raspiblitz.info
/mnt/hdd/raspiblitz.conf
```

openoms, 2022-05-02:
I think the conclusion here is that the `blitz.datadrive.sh` needs a complete rewrite.

Probably should apply less logic and offer a simple menu to choose the intended disks or partitions in a menu if the defaults are not met.

ktecho, 2022-07-13:
> I think the conclusion here is that the `blitz.datadrive.sh` needs a complete rewrite.
> 
> Probably should apply less logic and offer a simple menu to choose the intended disks or partitions in a menu if the defaults are not met.

Yes, I fully agree. I'm trying to install this in a container inside a Proxmox server, and I cannot make it work because I don't have /dev/xxx:

```
Filesystem                       Size  Used Avail Use% Mounted on
/dev/mapper/vg-vm--103--disk--0   40G  9.5G   28G  26% /
/dev/mapper/vg-vm--103--disk--1  590G   56K  560G   1% /mnt/hdd
```

rootzoll, 2024-11-14:
With RaspiBlitz v1.11.3 installing on a Debian12 was optimized and tested. Closing issue.

closed 2024-11-14 by rootzoll