raspiblitz/raspiblitz #3083: No default pi user in new Raspberry Pi OS releases

HiLivin, 2022-05-01:
Since the new Raspberry Pi OS release from April 2022, one cannot simply ssh to their device with a default user `pi` and password `raspberry`.
**From now on there is no preconfigured `pi` user.**

Link to the article:
https://www.raspberrypi.com/news/raspberry-pi-bullseye-update-april-2022/

### The Workaround
 - use [Raspberry Pi Imager](https://www.raspberrypi.com/software/) and configure the username and password manually through the wizard

or

 - in the boot partition create another file named `userconf` (beside the empty `ssh` file) with the content as below:
```
pi:$6$Kz08VDDCgIAbrvHH$qhFcZsg2fqcemy9CsozcUQhIFmqlqPht4O1mDmu8t.gaP8dgyAn1CxRpeIIEKh0M4BbFSuG9vW.I6DnNmItgJ1
```
During the first boot a user `pi` with password `raspberry` will be created.
The above randomly-salted hash of the password can be produced with `echo 'raspberry' | openssl passwd -6 -stdin`.
It's in the form `$<6>$<salt>$<hash>`.

I suggest to update the instructions in README.md for those who wish to build the RaspiBlitz by themselves.

openoms, 2022-05-02:
As it is.made to build on multiple architectures and images including amd64 Debian, Ubuntu and Armbian there ia a check and fix for the pi user already: https://github.com/rootzoll/raspiblitz/blob/217c278e722f363576d52099115b1702cfa97b02/build_sdcard.sh#L329

rootzoll, 2022-05-02:
To be tested with new Raspberry Pi OS image

rootzoll, 2022-06-01:
Tests look good ... closing for v1.8.0 release.

closed 2022-06-01 by rootzoll