raspiblitz/raspiblitz #3097: CLRest is not installed from connect menu

openoms, 2022-05-07:
needs to be installed when cRTL is not already present https://github.com/rootzoll/raspiblitz/blob/55215299e1f9ab6c10e09b255594d4c3e36c176b/home.admin/97addMobileWallet.sh#L284

rootzoll, 2022-05-15:
@openoms is clrest already part of the basic install of core lightning so that the BlitzAPI can use it? Or does its still need manual install by user?

openoms, 2022-05-15:
it is still a manual install with cRTL or now from the CONNECT menu also.
Would make sense to use it as part of the fatpack install with CLN. The c-lightningREST is still more functional than the cln-grpc plugin as far as I could see.

rootzoll, 2022-05-15:
@fusion44 you need the cln-grpc for BlitzAPI, not the c-lightningREST API, correct?

fusion44, 2022-05-15:
Correct 🙂 you can find a description on what is necessary to activate it here: https://github.com/jamaljsr/polar/issues/556

rootzoll, 2022-05-15:
@openoms ah sorry I got confused with core lightnbing APIs for a moment here. So the c-lightningREST API is not needed for BlitzAPI and its OK to have them just manually installed for extra apps. Do we have the cln-grpc API ready so far to build a first test image for the v1.8.0 with the BlitzAPI?

openoms, 2022-05-15:
I have only tried the installation so far, that is working. The plugin is running and the port is open (exposed over Tor as well). Don't know yet how to interact with it, but can ask.

fusion44, 2022-05-15:
If you want to try you can use the [cln branch of the API](https://github.com/fusion44/blitz_api/tree/cln). Almost all of the calls are implemented. 

Some crucial things are still missing from the CLN side. For example there is no way to decode a bolt11 String or fund a channel through gRPC. I'd like to avoid having another dependency just to decode at bolt11 string. Some features might have to wait until 0.12 of CLN is released. 

I hope that 0.12 also includes the accountability plugin so we don't have to read data from CLNs SQlite database.

commit 14d1ba7c (2022-05-13): install clrest on connect if not active, fix #3097
closed 2022-07-27 by rootzoll