raspiblitz/raspiblitz #3103: Failure after installing Touchdisplay

TAP2BTC, 2022-05-10:
Hey guys, since two weeks i'm running my own node with a raspiblitz. A few days ago i was playing a Little Bit with it and installed the touchdisplay (didnt see the Info here in github) and now i don’t know what to do. I uninstalled the touchdisplay and rebooted my blitz for two times but i still see the Display like in the Photo (now without the touchbuttons/touchdisplay). 
[image]
[image]
My infodisplay looks like the second Photo.
I‘ve my debug logs which you see bellow. 
I can only Go back to the Mainmenu with ./00mainMenu.sh. 
I’m new in this and an absolute beginner and i don’t know what to do with this. Can anyone Tell me what i have to do know? I can‘t find something like this anywhere. 

* RASPIBLITZ DEBUG LOGS 
***************************************************************
blitzversion: 1.7.2
chainnetwork: bitcoin / main
 22:39:57 up 18 min,  2 users,  load average: 2.18, 2.26, 1.69

*** SETUPPHASE / BOOTSTRAP ***
see logs: cat /home/admin/raspiblitz.log
setupPhase--> done
state--> ready

*** BACKGROUNDSERVICE ***
to monitor Background service call: sudo journalctl -f -u background

*** BLOCKCHAIN (MAINNET) SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
     Active: activating (start) since Mon 2022-05-09 22:39:43 BST; 14s ago
    Process: 135019 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
    Process: 135035 ExecStartPre=/bin/chgrp bitcoin /mnt/hdd/bitcoin (code=exited, status=0/SUCCESS)
Cntrl PID: 135036 (bitcoind)
      Tasks: 14 (limit: 4532)
        CPU: 14.021s
     CGroup: /system.slice/bitcoind.service
             ├─135036 /usr/local/bin/bitcoind -daemonwait -pid=/mnt/hdd/bitcoin/bitcoind.pid -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin -debuglogfile=/mnt/hdd/bitcoin/debug.log
             └─135069 /usr/local/bin/bitcoind -daemonwait -pid=/mnt/hdd/bitcoin/bitcoind.pid -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin -debuglogfile=/mnt/hdd/bitcoin/debug.log

May 09 22:39:43 raspberrypi systemd[1]: bitcoind.service: Consumed 16.374s CPU time.
May 09 22:39:43 raspberrypi systemd[1]: Starting Bitcoin daemon...

*** LAST BLOCKCHAIN (MAINNET) ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Journal begins at Fri 2022-05-06 19:54:30 BST, ends at Mon 2022-05-09 22:39:57 BST. --
May 09 22:39:41 raspberrypi systemd[1]: bitcoind.service: Control process exited, code=killed, status=11/SEGV
May 09 22:39:42 raspberrypi systemd[1]: bitcoind.service: Failed with result 'signal'.
May 09 22:39:42 raspberrypi systemd[1]: Failed to start Bitcoin daemon.
May 09 22:39:42 raspberrypi systemd[1]: bitcoind.service: Consumed 16.374s CPU time.
May 09 22:39:43 raspberrypi systemd[1]: bitcoind.service: Scheduled restart job, restart counter is at 59.
May 09 22:39:43 raspberrypi systemd[1]: Stopped Bitcoin daemon.
May 09 22:39:43 raspberrypi systemd[1]: bitcoind.service: Consumed 16.374s CPU time.
May 09 22:39:43 raspberrypi systemd[1]: Starting Bitcoin daemon...

*** LAST BLOCKCHAIN (MAINNET) 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2022-05-09T21:39:43Z Using /16 prefix for IP bucketing
2022-05-09T21:39:43Z Cache configuration:
2022-05-09T21:39:43Z * Using 2.0 MiB for block index database
2022-05-09T21:39:43Z * Using 8.0 MiB for chain state database
2022-05-09T21:39:43Z * Using 502.0 MiB for in-memory UTXO set (plus up to 286.1 MiB of unused mempool space)
2022-05-09T21:39:43Z init message: Loading block index…
2022-05-09T21:39:43Z Switching active chainstate to Chainstate [ibd] @ height -1 (null)
2022-05-09T21:39:43Z Opening LevelDB in /mnt/hdd/bitcoin/blocks/index
2022-05-09T21:39:43Z Opened LevelDB successfully
2022-05-09T21:39:43Z Using obfuscation key for /mnt/hdd/bitcoin/blocks/index: 0000000000000000
2022-05-09T21:39:54Z LoadBlockIndexDB: last block file = 3030
2022-05-09T21:39:54Z LoadBlockIndexDB: last block file info: CBlockFileInfo(blocks=9, size=13925808, heights=735506...735514, time=2022-05-08...2022-05-08)
2022-05-09T21:39:54Z Checking all blk files are present...
2022-05-09T21:39:54Z Opening LevelDB in /mnt/hdd/bitcoin/chainstate
2022-05-09T21:39:54Z Opened LevelDB successfully
2022-05-09T21:39:54Z Using obfuscation key for /mnt/hdd/bitcoin/chainstate: 39341d2937aed7b8
2022-05-09T21:39:55Z Loaded best chain: hashBestChain=00000000000000000006918a850b1af41497024b7ed333c16ffda77db0b0d9a2 height=735514 date=2022-05-08T20:27:51Z progress=0.999700
2022-05-09T21:39:55Z init message: Verifying blocks…
[... 146 lines cut ...]
system_ram_gb="4"
system_board="rp4"
system_count_start_lightning="0"
system_ups_status="OFF"
system_count_start_tui="0"
ln_default_error_short=""
ln_default_online="0"
ln_default_version="0.14.2-beta"
ln_default_error_full=""
ln_default_running="0"
ln_default_locked="0"
ln_default_activated="1"
ln_default_ready="0"
ln_default_sync_initial_done="1"
btc_default_error_full=""
btc_default_error_short=""
btc_default_online="0"
btc_default_ready="0"
btc_default_running="0"
btc_default_sync_initial_done="1"
btc_default_activated="1"
btc_default_version="v22.0.0"
_____________________________________
Will be very thankful for some help and sure i‘m thankful for your work you do for all of us! 🙏🏼
Andy

openoms, 2022-05-10:
Enter the menu and disable the Touchscreen in SETTINGS. It is not working well for now.

TAP2BTC, 2022-05-10:
Yes i done that. But now my Display shows this in the picture after every reboot/Start of my Blitz. Also every time I exit Settings or other points in the main menu, I have to go over ./00mainMenu.sh. Otherwise i only see a black Display. 
[image]

TAP2BTC, 2022-05-31:
I flashed the SD-Card & built a new one. Everything is Fine now. Merci

closed 2022-05-31 by TAP2BTC