raspiblitz/raspiblitz #3104: c-lightning starts, but doesn't function

Gilbog, 2022-05-10:
I run only c-lightning on Raspiblitz 1.7.2 on a classic Raspberry Pi setup.

I think I messed something up with my c-lightning channels trying to back up my channels. 

Here is what I did: 

- I tried running a backup as instructed on https://bitcointalk.org/index.php?topic=5384133.0. It worked fine the first time, created a backup.bkp file on an attached USB
- Today I tried creating the backup.bkp file again because I saw it hadn't updated after some channel changes. When I did that, I had forgotten to stop c-lightning before running `plugin/backup/backup-cli init --lightning-dir /home/bitcoin/.lightning/bitcoin/ file:///mnt/static-channel-backup-CLN-external/backup/backup.bkp`. The backup seemed to run successful.
- When I realized my mistake, I redid the backup, this time actually stopping the lightningd service first
- Restarting the lingtningd service seemed fine, and `sudo service lightningd status` says it's active
- But now my channels are not showing on the INFO screen. It says "C-LIGHTNING v0.10.2  Not Started | Not Ready Yet | No Data" for a long time already

Can somebody help me fix what I screwed up here? There are a lot of sats in those channels ...

openoms, 2022-05-10:
run:
`config.scripts/cl-plugin.backup.sh on`

the backup plugin is running by default and syncs to the SDcard. No need to configure manually.

Gilbog, 2022-05-10:
Thanks. I just ran that command. But the Info screen still says "C-LIGHTNING v0.10.2   Not Started | Not Ready Yet | No Data"

Anything else I need to do to get the channels/funds back?

openoms, 2022-05-10:
run:
`sudo systemctl restart lightningd`

see the logs in the SYSTEM menu and check:
`sudo journalctl -fu lightningd -n 30`

Gilbog, 2022-05-10:
Hm, lots of "FAILURE" on the log: 
...
May 10 17:53:21 raspberrypi systemd[1]: lightningd.service: Failed with result 'exit-code'.
May 10 17:53:21 raspberrypi systemd[1]: lightningd.service: Consumed 3.106s CPU time.
May 10 17:53:51 raspberrypi systemd[1]: lightningd.service: Scheduled restart job, restart counter is at 1.
May 10 17:53:51 raspberrypi systemd[1]: Stopped c-lightning daemon on mainnet.
May 10 17:53:51 raspberrypi systemd[1]: lightningd.service: Consumed 3.106s CPU time.
May 10 17:53:51 raspberrypi systemd[1]: Starting c-lightning daemon on mainnet...
May 10 17:53:51 raspberrypi systemd[1]: Started c-lightning daemon on mainnet.
May 10 17:53:53 raspberrypi sh[1611233]: Could not initialize the plugins, see above for details.
May 10 17:53:53 raspberrypi systemd[1]: lightningd.service: Main process exited, code=exited, status=1/FAILURE
May 10 17:53:53 raspberrypi systemd[1]: lightningd.service: Failed with result 'exit-code'.
May 10 17:53:53 raspberrypi systemd[1]: lightningd.service: Consumed 3.161s CPU time.
May 10 17:54:23 raspberrypi systemd[1]: lightningd.service: Scheduled restart job, restart counter is at 2.
May 10 17:54:23 raspberrypi systemd[1]: Stopped c-lightning daemon on mainnet.
May 10 17:54:23 raspberrypi systemd[1]: lightningd.service: Consumed 3.161s CPU time.
May 10 17:54:23 raspberrypi systemd[1]: Starting c-lightning daemon on mainnet...
May 10 17:54:24 raspberrypi systemd[1]: Started c-lightning daemon on mainnet.
May 10 17:54:26 raspberrypi sh[1614561]: Could not initialize the plugins, see above for details.
May 10 17:54:26 raspberrypi systemd[1]: lightningd.service: Main process exited, code=exited, status=1/FAILURE
May 10 17:54:26 raspberrypi systemd[1]: lightningd.service: Failed with result 'exit-code'.
May 10 17:54:26 raspberrypi systemd[1]: lightningd.service: Consumed 3.199s CPU time.

Gilbog, 2022-05-10:
I hadn't known that there is already a backup function in Raspiblitz. I had thought *I* must install some backup mechanism, so I followed https://bitcointalk.org/index.php?topic=5384133.0. 

Relevant to your question, I followed these instructions:

"Add plugin to lightningd.conf config file.
`nano /home/bitcoin/.lightning/bitcoin/lightningd.conf`
Simply add a new line with:
important-plugin=/home/bitcoin/plugins/backup/backup.py"

Should I remove that line again and run any more commands?

openoms, 2022-05-10:
>"Add plugin to lightningd.conf config file.
nano /home/bitcoin/.lightning/bitcoin/lightningd.conf
Simply add a new line with:
important-plugin=/home/bitcoin/plugins/backup/backup.py"

delete this whole file and restart lightningd. 

The config file on the raspiblitz is:
`/home/bitcoin/.lightning/bitcoin/config`

Gilbog, 2022-05-10:
Yes, it worked, THANK YOU!

One more question about backups: 
Does Raspiblitz automatically back up channels in c-lightning or do I have to do anything manually?
I saw there is a backup entry on CL > REPAIR-CL > BACKUP. Do I have to trigger this occasionally (at a channel change?)

openoms, 2022-05-10:
As above, the backup plugin does continously save to the SDcard. There is no intermittent backup option like the SCB in LND.

closed 2022-05-10 by openoms

openoms, 2022-05-10:
See the backup file - which can be restored manually only for now - in
```
/home/bitcoin/
```

Gilbog, 2022-05-10:
One more minor problem: The channels are visible and functioning in RTL or on the terminal. But the small monitor _on the Raspiblitz itself_ still shows "C-LIGHTNING v0.10.2 Not Started | Not Ready Yet | No Data" -- while the INFO screen from the main menu, however, shows the channels as usual

I tried rebooting, but that didn't fix it. 

How can I get the monitor on the Raspiblitz to update, too?