raspiblitz/raspiblitz #3105: WebUI over Tor

rootzoll, 2022-05-10:
Like mentioned in this comment its the goal to promote the WebUI also over Tor:
https://github.com/rootzoll/raspiblitz/issues/2482#issuecomment-924274178

This should be revisited for v1.8.1 - as for v1.8.0 its just the focus to provide the WebUI with plain http in the local network.

cstenglein, 2024-01-11:
Is the tor address published on the setup?
To check @rootzoll

cstenglein, 2024-09-26:
Ping @rootzoll for my question

> Is the tor address published on the setup?

rootzoll, 2024-09-26:
> Is the tor address published on the setup?

No and I think it is not a good idea to use Tor during setup. Setup is normaly happening in the local network (mostly behind NAT firewall) and local IP is fine for that. It makes more sense to offer a self-signed cert for HTTPS for people that like to have a secure communication even in their LAN. But not Tor for Setup - its often too slow and exposes a fresh RaspiBlitz with default password to the outside.

Only after Setup to reach the WebUI it makes sense to provide Tor.

cstenglein, 2024-10-04:
This would need a extra API endpoint.

The tor address currently comes from the `/system/get-system-info` endpoint, which needs authentication.

To offer Tor before login, I need an endpoint which does not need auth which just gives me the tor address or none if no tor is configured.