raspiblitz/raspiblitz #3110: WebUI ElectrumServer as an App

rootzoll, 2022-05-11:
Even if electrum server (electrs) has no WebUI itself - it makes to offer its install soon after the v1.8.0 release thru the WebUI in the App section - showing status and connection options. So that people can use it with their own electrum apps and hardware wallets.

rootzoll, 2022-12-14:
Sadly needs to be pished to v1.10 because of missing time.

cstenglein, 2023-11-16:
Is there any info provided from raspiblitz / the api for electrs already?

Got the basic work for the ui part done, just needs some interaction ;)

rootzoll, 2023-11-23:
I have started to prepare the electrs script for webinstall - Work in Progress

rootzoll, 2023-12-12:
Syndata can be gathered from the following script:
`/home/admin/config.scripts/bonus.electrs.sh status-sync`

rootzoll, 2023-12-12:
More Application Status of electrs could be called with: `/home/admin/config.scripts/bonus.electrs.sh status` but that should not be called in monitoring loop

rootzoll, 2024-03-19:
Works on my test - closing issue for release.

closed 2024-03-19 by rootzoll