raspiblitz/raspiblitz #3111: Expose & Manage Bitcoin RPC easy thru WebUI

rootzoll, 2022-05-11:
At the moment the `99connectMenu.sh` is containing management of Bitcoins RPC connection mixed with SSH-UI elements. It makes sense to seperate the RPC config & status out into a signe script that then also can be triggered by BlitzAPI to cater to the WebUI.

Also it should reflect more compexity like providing an `update` action that a bitcoind serbvice can call in prestart to  update local network binds to maybe changed localIP and local network configuration. Also an `provision` action that on update/recover can take care of ports being reopen etc - to keep settings stable over updates.

Here is a first draft of a `bitcoin.rpc.sh` config script - to be continued ...

```
if [ $# -eq 0 ] || [ "$1" = "-h" ] || [ "$1" = "-help" ]; then
  echo "Manage Bitcoin RPC"
  echo "bitcoin.rpc.sh [mainnet|testnet|signet] status [?force-activate-all]"
  echo "bitcoin.rpc.sh [mainnet|testnet|signet] local [on|off]"
  echo "bitcoin.rpc.sh [mainnet|testnet|signet] tor [on|off]"
  echo "bitcoin.rpc.sh provision --> use on update/recover to open needed etc"
  echo "bitcoin.rpc.sh update --> use to update localIP network changes (use on bitcoind prestart)"
  echo
  exit 1
fi

# 1. network selection [mainnet|testnet|signet]
if [ "$1" == "mainnet" ]; then
    chain="mainnet"
    chain_prefix=""
    rpcport="8332"
elif [ "$1" == "testnet" ]; then
    chain="testnet"
    chain_prefix="t"
    rpcport="18332"
elif [ "$1" == "signet" ]; then
    chain="signet"
    chain_prefix="s"
    rpcport="38332"
else
    echo "error='unkown network selection'"
    exit 1
fi

# Status on Bitcoin RPC connections

if [ "$2" == "status" ]; then

    rpcUser=$(sudo cat /mnt/hdd/bitcoin/bitcoin.conf | grep rpcuser | cut -c 9-)
    rpcPass=$(sudo cat /mnt/hdd/bitcoin/bitcoin.conf | grep rpcpassword | cut -c 13-)

fi

# Bitcoin RPC over Local Network

if [ "$2" == "local" ]; then
    if [ "$3" == "on" ]; then
        echo "# bitcoin.rpc.sh --> OPEN LOCAL NETWORK for ${chain}"

        # get local network information
        source <(/home/admin/_cache.sh get internet_localip internet_localiprange)
        localIP="${internet_localip}"
        localIPrange="${internet_localiprange}"

        # switch on answering RPC on the local IP of the raspiblitz
        bindIP=$(grep -c "${chain}.rpcbind=$localIP" <  /mnt/hdd/bitcoin/bitcoin.conf)
        if [ $bindIP -eq 0 ]; then
          echo "${chain}.rpcbind=$localIP" | sudo tee -a /mnt/hdd/bitcoin/bitcoin.conf
        fi

        # allow local IP access
        allowIPrangeExists=$(grep -c "^rpcallowip=" <  /mnt/hdd/bitcoin/bitcoin.conf)
        if [ $allowIPrange -eq 0 ]; then
          echo "rpcallowip=$localIPrange" | sudo tee -a /mnt/hdd/bitcoin/bitcoin.conf
        fi

        echo "# ufw allow from $localIPrange to any port ${BITCOINRPCPORT}"
        sudo ufw allow from $localIPrange to any port ${BITCOINRPCPORT}

        echo "# TODO"
        exit 0
    elif [ "$3" == "off" ]; then
        echo "# bitcoin.rpc.sh --> CLOSE LOCAL NETWORK for ${chain}"
        echo "# TODO"
        exit 0
    else
        echo "error='unkown switch'"
        exit 1
    fi
fi
    
# Bitcoin RPC over Tor Network

if [ "$2" == "tor" ]; then
    if [ "$3" == "on" ]; then
        echo "# bitcoin.rpc.sh --> OPEN TOR NETWORK for ${chain}"
        echo "# TODO"

    rpcTorService=$(grep -c "HiddenServicePort ${rpcport} 127.0.0.1:${rpcport}"  < /etc/tor/torrc)
    TorRPCaddress=$(sudo cat /mnt/hdd/tor/bitcoin${rpcport}/hostname)

        exit 0
    elif [ "$3" == "off" ]; then
        echo "# bitcoin.rpc.sh --> CLOSE TOR NETWORK for ${chain}"
        echo "# TODO"
        exit 0
    else
        echo "error='unkown switch'"
        exit 1
    fi
fi

echo "error='unkown action'"
exit 1
```

cstenglein, 2023-11-15:
Looks like there would be some work on the API side required as well right @rootzoll ?

Tagging this also with API, please remove if wrong.

rootzoll, 2023-11-15:
YEah seems a bit more work - pushing to a bit later release.

rootzoll, 2024-09-23:
@rootzoll make a requirement list/concept
@fusion44 let me know if you can fut in the API side on the