raspiblitz/raspiblitz #3114: shutdown / reboot leads to "unexpected power loss" on SSD

PatrickScheich, 2022-05-12:
After i had lost a raspiblitz instance to a failing HDD and my last SSD had shown some unrecoverable read errors.

I installed the smartmon-tools on my raspiblitz and tried to figure out if there are any metrics, obtainable via `smartctl -a /dev/sda`, which would help to determine the health status of my SSD.

```
root~# smartctl -a /dev/sda

smartctl 7.2 2020-12-30 r5155 [aarch64-linux-5.10.92-v8+] (local build)
Copyright (C) 2002-20, Bruce Allen, Christian Franke, www.smartmontools.org

=== START OF INFORMATION SECTION ===
Model Family:     Marvell based SanDisk SSDs
Device Model:     SanDisk SSD PLUS 1000GB
Serial Number:    211702A00D3C
LU WWN Device Id: 5 001b44 8babd6476
Firmware Version: UH4504RL
User Capacity:    1,000,204,886,016 bytes [1.00 TB]
Sector Size:      512 bytes logical/physical
Rotation Rate:    Solid State Device
Form Factor:      2.5 inches
TRIM Command:     Available, deterministic
Device is:        In smartctl database [for details use: -P show]
ATA Version is:   ACS-3, ACS-2 T13/2015-D revision 3
SATA Version is:  SATA 3.2, 6.0 Gb/s (current: 6.0 Gb/s)
Local Time is:    Thu May 12 02:07:43 2022 CEST
SMART support is: Available - device has SMART capability.
SMART support is: Enabled

=== START OF READ SMART DATA SECTION ===
SMART overall-health self-assessment test result: PASSED

General SMART Values:
Offline data collection status:  (0x00) Offline data collection activity
                                        was never started.
                                        Auto Offline Data Collection: Disabled.
Self-test execution status:      (   0) The previous self-test routine completed
                                        without error or no self-test has ever
                                        been run.
Total time to complete Offline
data collection:                (  120) seconds.
Offline data collection
capabilities:                    (0x11) SMART execute Offline immediate.
                                        No Auto Offline data collection support.
                                        Suspend Offline collection upon new
                                        command.
                                        No Offline surface scan supported.
                                        Self-test supported.
                                        No Conveyance Self-test supported.
                                        No Selective Self-test supported.
SMART capabilities:            (0x0003) Saves SMART data before entering
                                        power-saving mode.
                                        Supports SMART auto save timer.
Error logging capability:        (0x01) Error logging supported.
                                        General Purpose Logging supported.
Short self-test routine
recommended polling time:        (   2) minutes.
Extended self-test routine
recommended polling time:        ( 182) minutes.

SMART Attributes Data Structure revision number: 1
Vendor Specific SMART Attributes with Thresholds:
ID# ATTRIBUTE_NAME          FLAG     VALUE WORST THRESH TYPE      UPDATED  WHEN_FAILED RAW_VALUE
  5 Reallocated_Sector_Ct   0x0032   100   100   000    Old_age   Always       -       0
  9 Power_On_Hours          0x0032   100   100   000    Old_age   Always       -       5557
 12 Power_Cycle_Count       0x0032   100   100   000    Old_age   Always       -       23
165 Total_Write/Erase_Count 0x0032   100   100   000    Old_age   Always       -       1089
166 Min_W/E_Cycle           0x0032   100   100   ---    Old_age   Always       -       6
167 Min_Bad_Block/Die       0x0032   100   100   ---    Old_age   Always       -       98
168 Maximum_Erase_Cycle     0x0032   100   100   ---    Old_age   Always       -       24
169 Total_Bad_Block         0x0032   100   100   ---    Old_age   Always       -       583
170 Unknown_Marvell_Attr    0x0032   100   100   ---    Old_age   Always       -       0
171 Program_Fail_Count      0x0032   100   100   000    Old_age   Always       -       0
172 Erase_Fail_Count        0x0032   100   100   000    Old_age   Always       -       0
173 Avg_Write/Erase_Count   0x0032   100   100   000    Old_age   Always       -       6
174 Unexpect_Power_Loss_Ct  0x0032   100   100   000    Old_age   Always       -       13
184 End-to-End_Error        0x0032   100   100   ---    Old_age   Always       -       0
187 Reported_Uncorrect      0x0032   100   100   000    Old_age   Always       -       0
188 Command_Timeout         0x0032   100   100   ---    Old_age   Always       -       0
194 Temperature_Celsius     0x0022   045   063   000    Old_age   Always       -       55 (Min/Max 20/63)
199 SATA_CRC_Error          0x0032   100   100   ---    Old_age   Always       -       0
230 Perc_Write/Erase_Count  0x0032   100   100   000    Old_age   Always       -       779 276 779
232 Perc_Avail_Resrvd_Space 0x0033   100   100   005    Pre-fail  Always       -       100
233 Total_NAND_Writes_GiB   0x0032   100   100   ---    Old_age   Always       -       6425
234 Perc_Write/Erase_Ct_BC  0x0032   100   100   000    Old_age   Always       -       14707
241 Total_Writes_GiB        0x0030   100   100   000    Old_age   Offline      -       5486
242 Total_Reads_GiB         0x0030   100   100   000    Old_age   Offline      -       59842
244 Thermal_Throttle        0x0032   000   100   ---    Old_age   Always       -       0

SMART Error Log Version: 1
No Errors Logged

SMART Self-test log structure revision number 1
Num  Test_Description    Status                  Remaining  LifeTime(hours)  LBA_of_first_error
# 1  Extended offline    Completed without error       00%         4         -
# 2  Extended offline    Aborted by host               90%         1         -
# 3  Short offline       Completed without error       00%         1         -

Selective Self-tests/Logging not supported
```
After monitoring these for some weeks. I found that the values of attributes `12 Power_Cycle_Count` and `174 Unexpect_Power_Loss_Ct` increased whenever i rebooted or powered-off my raspiblitz, using the menu via ssh.

As a test I loged in as root via ssh and issued the `systemctl stop ...` and `shutdown -r now` commands manually.

This lead to an increase of  `12 Power_Cycle_Count` but no change in`174 Unexpect_Power_Loss_Ct` 

I will try to find a way to reboot / shutdown / power-off the raspiblitz via menue, which does not lead to an unexpeted powerloss for the SSD.

PatrickScheich, 2022-05-12:
> As a test I loged in as root via ssh and issued the `systemctl stop ...` and `shutdown -r now` commands manually.
> This lead to an increase of `12 Power_Cycle_Count` but no change in `174 Unexpect_Power_Loss_Ct`

unfortunately I can not reproduce this....the  `174 Unexpect_Power_Loss_Ct` increases with each reboot / power-off indepentant if it has benn initiated via menue or in ssh session

PatrickScheich, 2022-05-18:
After conducting some test, the situation seems as follows (at least for my Hardware / Software setup)  

The SSD experience an ***"Unexpected Powerloss"*** if `shutdown -r now`  or `shutdown -h now` is used.  
The only way to prevent this, is using `shutdown -P now` ... wait for some secondes and manualy powercycle the RaspiBlitz.

I would like to encourage the comunity to contuct thier own tests as different Hardware setups SSD + USB-to-SATA adaptors may behave different.

rootzoll, 2022-05-18:
Thanks for the research on this. Which USB-to-SATA controller are you using? The UGREEN as linked in the RaspiBlitz shopping list: https://github.com/rootzoll/raspiblitz#amazon-shopping-list-buy-parts--build-it-yourself

PatrickScheich, 2022-05-19:
> Thanks for the research on this. Which USB-to-SATA controller are you using? The UGREEN as linked in the RaspiBlitz shopping list: https://github.com/rootzoll/raspiblitz#amazon-shopping-list-buy-parts--build-it-yourself

Yes, exactly

```
root@DesertRetreatLN:~# lsusb -vvv -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    ID 1d6b:0003 Linux Foundation 3.0 root hub
    /sys/bus/usb/devices/usb2  /dev/bus/usb/002/001
    |__ Port 2: Dev 2, If 0, Class=Mass Storage, Driver=uas, 5000M
        ID 174c:1153 ASMedia Technology Inc. ASM1153 SATA 3Gb/s bridge
        /sys/bus/usb/devices/2-2  /dev/bus/usb/002/002
```

openoms, 2023-12-02:
We could add a systemd service which runs when shutdown or reboot is triggered like:

```
[Unit]
Description=Clean Shutdown Service
DefaultDependencies=no
Before=halt.target shutdown.target reboot.target

[Service]
Type=oneshot
ExecStart=/home/admin/config.scripts/blitz.shutdown.sh
RemainAfterExit=yes

[Install]
WantedBy=halt.target shutdown.target reboot.target
```

from: https://askubuntu.com/questions/1162420/how-to-run-commands-on-shutdown-with-systemd

openoms, 2023-12-04:
Interesting details connected to the X825 SATA-USB board:

the Samsung 870 QVO disk does not have the `174 Unexpect_Power_Loss_Ct` metric while the Sandisk SSD Plus and WD Blue disks have it.