raspiblitz/raspiblitz #3115: Problems updating from 1.7.1 to 1.7.2: stuck at "LND Not Started | Not Ready Yet"

gittbuh, 2022-05-12:
Hello,

I am trying to update my node from 1.7.1 to 1.7.2 following the usual method, following the update option from the main menu. Whether I choose to make a backup to my computer or not, when I click to proceed, so that the node can be switched off and the memory card replaced with the newer version, it gets stuck with the following red message at the bottom of the screen: "LND Not Started | Not Ready Yet". I had to force-restart 3 times the node because of this. 

Thanks for any hints on what is going on.

openoms, 2022-05-13:
you can just switch off and boot with a new SDcard. v1.7.2 will compact the LND database which is likely the issue.

Can compact manually beforehand too: https://www.lightningnode.info/technicals/lnddatabasecompaction

cstenglein, 2023-06-07:
@gittbuh Do you still have this issue?

Closed for now, if it still appears, please re-open

closed 2023-06-07 by cstenglein