raspiblitz/raspiblitz #3116: Nextcloud SCB Script Error

nicklyk, 2022-05-12:
After filling out the login details to my nextcloud server, i get the following message:

```
Nextcloud Settings changed ..
rm: cannot remove '/var/cache/raspiblitz/.tmp': Operation not permitted
sed: -e expression #1, char 59: unknown option to `s'
err='Nextcloud credentials are missing'
```

nicklyk, 2022-05-20:
I got it to work! 

First, i added a `sudo` on line 39 in nextcloud.upload.sh
Also, when entering the server link, i had to escape the slashes of the link i had provided.

https://somelink.com ---> https:\/\/somelink.com

After that it worked flawlessly :+1:

rootzoll, 2022-05-30:
Thanks ... will see how I can improve the script.

rootzoll, 2022-06-02:
Improvements committed for v1.8.0 release. Tests looking good - closing issue.

closed 2022-06-02 by rootzoll