raspiblitz/raspiblitz #3120: [x86] failed to detect a appropriate volume for data

kieselbert, 2022-05-15:
```

admin@192.168.4.56:~₿ sudo lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE  MOUNTPOINTS
loop0         7:0    0  61,9M  1 loop  /snap/core20/1405
loop1         7:1    0  79,9M  1 loop  /snap/lxd/22923
loop2         7:2    0  44,7M  1 loop  /snap/snapd/15534
sda           8:0    0   1,8T  0 disk
└─md0         9:0    0   1,8T  0 raid1
sdb           8:16   0   1,8T  0 disk
└─md0         9:0    0   1,8T  0 raid1
nvme0n1     259:0    0 931,5G  0 disk
├─nvme0n1p1 259:1    0     1G  0 part  /boot/efi
└─nvme0n1p2 259:2    0 930,5G  0 part  /
```
```
Running Format: (ext4) (nvmen)
# DATA DRIVE - FORMATTING to EXT4 layout (old)
# given device was not found
# or is not of type disk - see 'lsblk'
FAIL ON FORMATTING THE DRIVE:
device not valid
Please report as issue on the raspiblitz github.
# menu loop received exit code 1 --> exit to terminal
```

```
admin@192.168.4.56:~₿ sudo config.scripts/blitz.datadrive.sh status
# RASPIBLITZ DATA DRIVE Status

# BASICS
isMounted=0
isBTRFS=0

# SETUP INFO
isSSD=0
hddCandidate='nvmen'
fdisk: cannot open /dev/nvmen: No such file or directory
(standard_in) 1: syntax error
hddBytes=
hddGigaBytes=
hddPartitionCandidate='nvme0n1'
hddDataPartitionBytes=1000204886016
hddPartitionGigaBytes=931
hddFormat='vfat
ext4'
hddRaspiData=0
hddBlocksBitcoin=0
hddGotBlockchain=0
hddAdapterUSB=''
hddAdapterUSAP=0

# RAID
isRaid=0
raidCandidate[0]='sda 1863 GB ATA'
raidCandidate[1]='sdb 1863 GB ATA'
raidCandidates=2

# SWAP
isSwapExternal=0

```

tried a few preparations

```
admin@192.168.4.56:~₿ sudo mkdir /mnt/hdd
admin@192.168.4.56:~₿ sudo mkfs /dev/md0 -t ext4
admin@192.168.4.56:~₿ sudo mount /dev/md0 /mnt/hdd
admin@192.168.4.56:~₿ sudo config.scripts/blitz.datadrive.sh status
# RASPIBLITZ DATA DRIVE Status

# BASICS
isMounted=1
isBTRFS=0
hddRaspiData=0
hddRaspiVersion=''
isSSD=0
datadisk='md'
datapartition='md0'
hddCandidate='md'
hddPartitionCandidate='md0'
hddBlocksBitcoin=0
hddBytes=2000263643136
2000263643136
hddGigaBytes=2000263643136
1862
hddUsedInfo='11G
0
5,3M
28K (2011%)'
config.scripts/blitz.datadrive.sh: line 430: * 1024: syntax error: operand expected (error token is "* 1024")
error='unkown command'

```

that error in line 430 also appears when I use a single sda drive (instead of raid)

kieselbert, 2022-05-16:
my workaround:
1. prepared a volume 
sudo mkfs /dev/sdb -t ext4
sudo mount /dev/sdb /mnt/hdd
2. correct cached values
sudo /home/admin/_cache.sh get hddCandidate  --> hddCandidate="nvmen"
sudo /home/admin/_cache.sh set hddCandidate  sdb 
3. litte change in script
vi /home/admin/config.scripts/blitz.datadrive.sh 
Line 93
lsblk -o NAME,SIZE -b | grep -P "[s|v**n**][dv][a-z][0-9]?" > .lsblk.tmp
changed to
lsblk -o NAME,SIZE -b | grep -P "[s|v][dv][a-z][0-9]?" > .lsblk.tmp

Please build in an "advanced mode" where users can select which drive they want to use.

rootzoll, 2022-12-14:
@openoms is this still an issue?

rootzoll, 2025-05-15:
v1.12.0 will come with a complete new blitz.data.sh for drive discovery. Closing issue.