raspiblitz/raspiblitz #3121: [x86] Set passwords (A,B,C) not possible during setup

kieselbert, 2022-05-16:
During the installation process there is a point where I should define the passwords A, B & C. But I dont get the input fields, it instandly says password accepted and I only can press ok. This happens for every password.
After that the installation failed, thats the ouput:

```
###########################################################
# /home/admin/raspiblitz.log

***********************************************
Running RaspiBlitz Bootstrap 1.8.0rc1
Mo 16. Mai 11:45:55 UTC 2022
***********************************************
# *** /home/admin/config.scripts/blitz.ssh.sh checkrepair
# countKeyFiles(8)
# OK: SSHD RUNNING
## INIT raspiblitz.info
baseimage=ubuntu
cpu=x86_64
displayClass=headless
displayType=
setupPhase=boot
setupStep=0
fsexpanded=0
state=starting
btc_mainnet_sync_initial_done=0
btc_testnet_sync_initial_done=0
btc_signet_sync_initial_done=0
ln_lnd_mainnet_sync_initial_done=0
ln_lnd_testnet_sync_initial_done=0
ln_lnd_signet_sync_initial_done=0
ln_cl_mainnet_sync_initial_done=0
ln_cl_testnet_sync_initial_done=0
ln_cl_signet_sync_initial_done=0
## INIT RaspiBlitz Cache ... wait background.scan.service to finish first scan loop
- waiting for background.scan.service --> systemscan_runtime()
- waiting for background.scan.service --> systemscan_runtime()
- waiting for background.scan.service --> systemscan_runtime(2)
OK - logs are at 20 MB - within safety limit
Waiting for HDD/SSD ...
total 556
drwxr-xr-x   4 root root   4096 Mai 16 11:20 .
drwxr-xr-x 117 root root   4096 Mai 16 11:45 ..
-rw-r--r--   1 root root 505426 Feb 25 23:30 moduli
-rw-r--r--   1 root root   1650 Feb 25 23:30 ssh_config
drwxr-xr-x   2 root root   4096 Feb 25 23:30 ssh_config.d
-rw-r--r--   1 root root   3281 Mai 16 11:20 sshd_config
drwxr-xr-x   2 root root   4096 Feb 25 23:30 sshd_config.d
-rw-------   1 root root   1381 Mai 16 11:20 ssh_host_dsa_key
-rw-r--r--   1 root root    602 Mai 16 11:20 ssh_host_dsa_key.pub
-rw-------   1 root root    505 Mai 16 11:20 ssh_host_ecdsa_key
-rw-r--r--   1 root root    174 Mai 16 11:20 ssh_host_ecdsa_key.pub
-rw-------   1 root root    399 Mai 16 11:20 ssh_host_ed25519_key
-rw-r--r--   1 root root     94 Mai 16 11:20 ssh_host_ed25519_key.pub
-rw-------   1 root root   2602 Mai 16 11:20 ssh_host_rsa_key
-rw-r--r--   1 root root    566 Mai 16 11:20 ssh_host_rsa_key.pub
-rw-r--r--   1 root root    342 Dez  7  2020 ssh_import_id
HDD/SSD connected: sdb
No HDMI switch found.
No FS EXPAND needed. needsExpansion(0) fsexpanded(0)
No WIFI RESTORE because no /var/cache/raspiblitz/hdd-inspect/wpa_supplicant.conf
No SSH SERVER CERTS RESTORE because no /var/cache/raspiblitz/hdd-inspect
No SSHRESET switch found.
No DISPLAY RESTORE because no /var/cache/raspiblitz/hdd-inspect/raspiblitz.conf
No UASP FIX needed
isMounted: 0
HDD already part of system: 0
Temp mounting (1) data drive
Temp mounting (1) result: 0
HDD is there but not AutoMounted yet - Waiting for user Setup/Update
# df
Filesystem     1K-blocks     Used Available Use% Mounted on
tmpfs            1571336     1572   1569764   1% /run
/dev/nvme0n1p2 411725224 10814792 379922528   3% /
tmpfs            7856676        0   7856676   0% /dev/shm
tmpfs               5120        0      5120   0% /run/lock
tmpfs              32768        0     32768   0% /var/cache/raspiblitz
/dev/nvme0n1p3 411725224   493968 390243352   1% /home
/dev/nvme0n1p1   1098632     5364   1093268   1% /boot/efi
# lsblk -o NAME,FSTYPE,LABEL
NAME        FSTYPE            LABEL
loop0       squashfs
loop1       squashfs
loop2       squashfs
sda         linux_raid_member btcnode:0
└─md0
sdb         ext4
nvme0n1
├─nvme0n1p1 vfat
├─nvme0n1p2 ext4
└─nvme0n1p3 ext4
# RASPIBLITZ DATA DRIVE Status

# BASICS
isMounted=0
isBTRFS=0

# SETUP INFO
isSSD=1
hddCandidate='sdb'
hddBytes=
hddGigaBytes=
hddPartitionCandidate=''
hddAdapterUSB=''
hddAdapterUSAP=0

# RAID
isRaid=0
raidCandidate[0]='sda 1863 GB ATA'
raidCandidates=1

# SWAP
isSwapExternal=0

Displaying Info Message: Please Login for Setup
## WAIT LOOP: USER SETUP/UPDATE/MIGRATION
# PROVISION PROCESS with setupPhase(setup)
# CREATING raspiblitz.conf from your setup choices
Temp mounting (2) data drive (hddFormat='ext4')
Temp mounting (2) result: 1
Refreshing links between directories/drives ..

*** Enable Tor Service ***
*** Create directories and set permissions ***
mkdir: created directory '/mnt/hdd/app-data/tor'
mkdir: created directory '/mnt/hdd/app-data/tor/torrc.d'
mkdir: created directory '/mnt/hdd/tor'
mkdir: created directory '/mnt/hdd/tor/sys'
mkdir: created directory '/mnt/hdd/tor/sys/keys'
mkdir: created directory '/mnt/hdd/tor/services'
mkdir: created directory '/mnt/hdd/tor/onion_auth'
mode of '/mnt/hdd/tor' changed from 0755 (rwxr-xr-x) to 0700 (rwx------)
mode of '/mnt/hdd/app-data/tor' retained as 0755 (rwxr-xr-x)
mode of '/mnt/hdd/app-data/tor/torrc.d' retained as 0755 (rwxr-xr-x)
mode of '/etc/tor/torrc' retained as 0644 (rw-r--r--)
failed to change mode of '/mnt/hdd/app-data/tor/torrc.d/*' from 0644 (rw-r--r--) to 0644 (rw-r--r--)
changed ownership of '/mnt/hdd/tor/sys/keys' from root:root to debian-tor:debian-tor
changed ownership of '/mnt/hdd/tor/sys' from root:root to debian-tor:debian-tor
changed ownership of '/mnt/hdd/tor/onion_auth' from root:root to debian-tor:debian-tor
changed ownership of '/mnt/hdd/tor/services' from root:root to debian-tor:debian-tor
changed ownership of '/mnt/hdd/tor' from root:root to debian-tor:debian-tor
ownership of '/mnt/hdd/app-data/tor/torrc.d' retained as root:root
ownership of '/mnt/hdd/app-data/tor' retained as root:root
- /etc/tor/torrc will get edited

*** updating Tor config /etc/tor/torrc ***
## raspiblitz torrc (anchor, do not remove this line)
## See 'man tor', or https://2019.www.torproject.org/docs/tor-manual.html.en
## See https://github.com/torproject/tor/blob/main/src/config/torrc.sample.in

%include /mnt/hdd/app-data/tor/torrc.d

DataDirectory /mnt/hdd/tor/sys
PidFile /mnt/hdd/tor/sys/tor.pid

SafeLogging 0
Log notice stdout
Log notice file /mnt/hdd/tor/notice.log
Log info file /mnt/hdd/tor/info.log

RunAsDaemon 1
ControlPort 9051
SocksPort 9050 IsolateDestAddr
ExitRelay 0
CookieAuthentication 1
CookieAuthFileGroupReadable 1

- /mnt/hdd/app-data/tor/torrc.d/bridges already edited
# DO NOT EDIT! This file is generated by raspiblitz and will be overwritten
[Service]
ReadWriteDirectories=-/mnt/hdd/tor
[Unit]
After=network.target nss-lookup.target mnt-hdd.mount

- add custom directories to apparmor
    # vim:syntax=apparmor
#include <tunables/global>

profile system_tor flags=(attach_disconnected) {
  #include <abstractions/tor>

  owner /var/lib/tor/** rwk,
  owner /var/lib/tor/ r,
  owner /var/log/tor/* w,

  owner /mnt/hdd/tor/** rwk,
  owner /mnt/hdd/tor/ r,
  owner /mnt/hdd/app-data/tor/** rwk,

  # During startup, tor (as root) tries to open various things such as
  # directories via check_private_dir().  Let it.
  /var/lib/tor/** r,
  /mnt/hdd/tor/** r,
  /mnt/hdd/app-data/tor/** r,

  /{,var/}run/tor/ r,
  /{,var/}run/tor/control w,
  /{,var/}run/tor/socks w,
  /{,var/}run/tor/tor.pid w,
  /{,var/}run/tor/control.authcookie w,
  /{,var/}run/tor/control.authcookie.tmp rw,
  /{,var/}run/systemd/notify w,

  # Site-specific additions and overrides. See local/README for details.
  #include <local/system_tor>
}

# df
Filesystem      1K-blocks     Used  Available Use% Mounted on
tmpfs             1571336     1596    1569740   1% /run
/dev/nvme0n1p2  411725224 10831256  379906064   3% /
tmpfs             7856676        0    7856676   0% /dev/shm
tmpfs                5120        0       5120   0% /run/lock
tmpfs               32768        8      32760   1% /var/cache/raspiblitz
/dev/nvme0n1p3  411725224   493976  390243344   1% /home
/dev/nvme0n1p1    1098632     5364    1093268   1% /boot/efi
tmpfs             1571332        4    1571328   1% /run/user/1000
/dev/sdb1      1921724608      112 1824032440   1% /mnt/hdd
# lsblk -o NAME,FSTYPE,LABEL
NAME        FSTYPE            LABEL
loop0       squashfs
loop1       squashfs
loop2       squashfs
sda         linux_raid_member btcnode:0
└─md0
sdb
└─sdb1      ext4              BLOCKCHAIN
nvme0n1
├─nvme0n1p1 vfat
├─nvme0n1p2 ext4
└─nvme0n1p3 ext4
# Sourcing /var/cache/raspiblitz/temp/raspiblitz.setup
# migrationFile()
# Sourcing /mnt/hdd/raspiblitz.conf
# RASPIBLITZ CONFIG FILE
raspiBlitzVersion='1.8.0rc1'
lcdrotate='1'
lightning='lnd'
network='bitcoin'
chain='main'
hostname='zireael'
runBehindTor='on'
# Sourcing /var/cache/raspiblitz/temp/raspiblitz.setup
***********************************************************
ERROR - please report to development team
***********************************************************
state(error) message(missing-passworda)
https://github.com/rootzoll/raspiblitz#support
command to shutdown --> off
admin@192.168.4.57:~₿
```

kieselbert, 2022-05-16:
ok, seems it is a v1.8rc1 issue
I switched back ti v1.7

rootzoll, 2022-12-14:
@openoms is this still an issue?

openoms, 2022-12-14:
had no reports since with a few people setting up new amd64 nodes. Closing.

closed 2022-12-14 by openoms