raspiblitz/raspiblitz #3122: Booting issue

TimmyP13, 2022-05-16:
Had a power outage and trying to boot back up my system. Seems to hang on this screen. Do I just need to wait longer or do I have an issue?
[image]
 Thanks!

rootzoll, 2022-05-17:
A power outage has always the danger of data corruption & loss. If you are lucky just your sd card with the OS eneded up ins some inconsistent sate. So first try should be to flash a new sd card with the latest RaspiBlitz image and see if you can get thru the recovery process. https://github.com/rootzoll/raspiblitz#downloading-the-software

TimmyP13, 2022-05-17:
Ok I will try it. Thanks for the response!

openoms, 2022-06-18:
please close if resolved or abandoned