raspiblitz/raspiblitz #3129: Running LNBits and LNURLs on Tor

seinzfiction, 2022-05-18:
Hi,
I've set up LNBits on my Raspiblitz 1.7.2 and its running on Tor. I wanted to make it accessible on the Clearnet via Reverse Proxy on a VPS as described here: https://docs.btcpayserver.org/Deployment/ReverseProxyToTor/ .
My LNBits instance was accessible via the reverse proxy without problems.
When I tried to use the LNBits LNURLp Extension to generate Payment URLs, I received the following error messages when accessing a generated LNURL Pay link:

INFO:     87.183.125.4:0 - "GET /lnurlp/api/v1/lnurl/11 HTTP/1.0" 200 OK
ERROR:    Exception in ASGI application
Traceback (most recent call last):
  File "/home/lnbits/lnbits/venv/lib/python3.9/site-packages/uvicorn/protocols/http/httptools_impl.py", line 375, in run_asgi
    result = await app(self.scope, self.receive, self.send)
  File "/home/lnbits/lnbits/venv/lib/python3.9/site-packages/uvicorn/middleware/proxy_headers.py", line 75, in __call__
    return await self.app(scope, receive, send)
  File "/home/lnbits/lnbits/venv/lib/python3.9/site-packages/fastapi/applications.py", line 208, in __call__
    await super().__call__(scope, receive, send)
  File "/home/lnbits/lnbits/venv/lib/python3.9/site-packages/starlette/applications.py", line 112, in __call__
    await self.middleware_stack(scope, receive, send)
  File "/home/lnbits/lnbits/venv/lib/python3.9/site-packages/starlette/middleware/errors.py", line 181, in __call__
    raise exc from None
  File "/home/lnbits/lnbits/venv/lib/python3.9/site-packages/starlette/middleware/errors.py", line 159, in __call__
    await self.app(scope, receive, _send)
  File "/home/lnbits/lnbits/venv/lib/python3.9/site-packages/starlette/middleware/gzip.py", line 18, in __call__
    await responder(scope, receive, send)
  File "/home/lnbits/lnbits/venv/lib/python3.9/site-packages/starlette/middleware/gzip.py", line 35, in __call__
    await self.app(scope, receive, self.send_with_gzip)
  File "/home/lnbits/lnbits/venv/lib/python3.9/site-packages/starlette/middleware/cors.py", line 78, in __call__
    await self.app(scope, receive, send)
  File "/home/lnbits/lnbits/venv/lib/python3.9/site-packages/starlette/exceptions.py", line 82, in __call__
    raise exc from None
  File "/home/lnbits/lnbits/venv/lib/python3.9/site-packages/starlette/exceptions.py", line 71, in __call__
    await self.app(scope, receive, sender)
  File "/home/lnbits/lnbits/venv/lib/python3.9/site-packages/starlette/routing.py", line 580, in __call__
    await route.handle(scope, receive, send)
  File "/home/lnbits/lnbits/venv/lib/python3.9/site-packages/starlette/routing.py", line 241, in handle
    await self.app(scope, receive, send)
  File "/home/lnbits/lnbits/venv/lib/python3.9/site-packages/starlette/routing.py", line 52, in app
    response = await func(request)
  File "/home/lnbits/lnbits/venv/lib/python3.9/site-packages/fastapi/routing.py", line 226, in app
    raw_response = await run_endpoint_function(
  File "/home/lnbits/lnbits/venv/lib/python3.9/site-packages/fastapi/routing.py", line 159, in run_endpoint_function
    return await dependant.call(**values)
  File "./lnbits/extensions/lnurlp/lnurl.py", line 34, in api_lnurl_response
    resp = LnurlPayResponse(
  File "/home/lnbits/lnbits/venv/lib/python3.9/site-packages/pydantic/main.py", line 406, in __init__
    raise validation_error
pydantic.error_wrappers.ValidationError: 3 validation errors for LnurlPayResponse
callback
  URL host invalid, top level domain required (type=value_error.url.host)
callback
  URL scheme not permitted (type=value_error.url.scheme; allowed_schemes={'https'})
  
I analyzed the communication. On my Raspiblitz machine the nginx service is listening for LNBits access based on the config file **/etc/nginx/sites-available/lnbits_tor.conf**. In the location section the entry **proxy_pass http://127.0.0.1:5000;** is passing the incoming Tor request via HTTP to the LNBits service. 
This causes the issue. I changed this line in the configuration file to: **proxy_pass https://127.0.0.1:5001** 
This forced to forward the incoming Tor request via HTTPS to the LNBits service.

After this change the LNURL Pay links were working like a charm.

From my point of view this configuration change should be part of one of the next Raspiblitz releases.

BitcoinQnA, 2022-05-18:
Can also confirm that this change fixed my issues too.

rootzoll, 2022-06-02:
@seinzfiction I am not completly shure why that works, because port 5001 will then ne also point to another nginx proxypass in the file `/etc/nginx/sites-available/lnbits_ssl.conf` that then goes to `http://127.0.0.1:5000` but I will give it a try for the next realease and when nobody reports further side effects keep it that way. Please test if this then works for you out of the box on v1.8.0rc2 release.

rootzoll, 2022-06-02:
Moving to v1.8.1 ro recheck for reported side effects before closing.

seinzfiction, 2022-06-04:
@rootzoll thank you for your feedback. I'll test it with the release candidate. Where can I find the 1.8.0rc2 release for testing?

rootzoll, 2022-06-06:
Preparing v1.8.0rc2 at the moment .. should come within this week.

closed 2022-12-15 by rootzoll