raspiblitz/raspiblitz #3133: Bitcoin Node Startup Issue

mroxso, 2022-05-19:
I restarted my raspiblitz (v1.7.2).

Now the bitcoin log shows the following:
`2022-05-19T07:40:25Z connect() to 127.0.0.1:9050 failed after wait: Connection refused (111)`

The bitcoin node is not booting up because of this.

openoms, 2022-05-20:
it is a Tor error. Monitor with:
`sudo journalctl -fu tor@default`

try:
`sudo systemctl restart tor`

or another full reboot.

mroxso, 2022-05-23:
mmh true. seems like my tor service is not working. but a restart isn't fixing it for me.

mroxso, 2022-05-23:
I fixed it with the following commands:
`sudo chmod 700 -R /mnt/hdd/tor/`
`sudo systemctl restart tor`

so i changed the permissions to be a bit more private.
i dont know why they were to peremissive.

now it seems to work fine again.

closed 2022-05-23 by mroxso