raspiblitz/raspiblitz #3134: Install Core Lightning & Bitcoin-only thru WebUI

rootzoll, 2022-05-19:
Weiting for the branch https://github.com/fusion44/blitz_api/tree/cln of the BlitzAPI to get merged and then test on the WebUI Setup & Dashboard https://github.com/cstenglein/raspiblitz-web/issues/289#

Also test if the AppStore reflects that c-lightning is running and not LND.

rootzoll, 2022-06-05:
When setting up a fresh core lightnig node the BackendAPI is not starting up - see:

```
Jun 05 16:51:49 raspberrypi systemd[1]: Starting BlitzBackendAPI...
Jun 05 16:51:50 raspberrypi blitz.web.api.sh[176446]: # Update Web API CONFIG (Sun  5 Jun 16:51:50 BST 2022)
Jun 05 16:51:50 raspberrypi sudo[176456]:     root : PWD=/root/blitz_api ; USER=root ; COMMAND=/usr/bin/cat /mnt/hdd/bitcoin/bitcoin.conf
Jun 05 16:51:50 raspberrypi sudo[176456]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=0)
Jun 05 16:51:50 raspberrypi sudo[176456]: pam_unix(sudo:session): session closed for user root
Jun 05 16:51:50 raspberrypi sudo[176475]:     root : PWD=/root/blitz_api ; USER=root ; COMMAND=/usr/bin/cat /mnt/hdd/bitcoin/bitcoin.conf
Jun 05 16:51:50 raspberrypi sudo[176475]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=0)
Jun 05 16:51:50 raspberrypi sudo[176475]: pam_unix(sudo:session): session closed for user root
Jun 05 16:51:50 raspberrypi blitz.web.api.sh[176446]: # CONFIG Web API Lightning --> CL
Jun 05 16:51:50 raspberrypi blitz.web.api.sh[176503]: xxd: /home/bitcoin/.lightning/bitcoin/client.pem: No such file or directory
Jun 05 16:51:50 raspberrypi blitz.web.api.sh[176504]: xxd: /home/bitcoin/.lightning/bitcoin/client-key.pem: No such file or directory
Jun 05 16:51:50 raspberrypi blitz.web.api.sh[176506]: xxd: /home/bitcoin/.lightning/bitcoin/ca.pem: No such file or directory
Jun 05 16:51:50 raspberrypi blitz.web.api.sh[176446]: # FAIL /home/bitcoin/.lightning/bitcoin/*.pem files maybe missing
Jun 05 16:51:50 raspberrypi blitz.web.api.sh[176446]: # '.env' config updates - blitzapi maybe needs to be restarted
Jun 05 16:51:50 raspberrypi systemd[1]: Started BlitzBackendAPI.
```

rootzoll, 2022-06-05:
OK `/home/admin/config.scripts/cl-plugin.cln-grpc.sh on` was missing on provision when core lightning was selected.

rootzoll, 2022-06-05:
OK I got the API running for core lightning fresh node ... but like with bitcoin-only mode the WebUI can now login but just see no data coming from the api like in this issue: #3135

@fusion44 should we run the API with `cln_unix_socket` or `cln_grpc`  ... at the moment I set it to `cln_grpc`

fusion44, 2022-06-06:
Please use the GRPC version. I'll probably remove the Unix Socket variant as it is incomplete and slow.

rootzoll, 2022-06-17:
OK tested & fixed core lightning & bitcoin-only. Closing issue for v1.8.0 RC2 release.

closed 2022-06-17 by rootzoll