raspiblitz/raspiblitz #3135: API+WebUI: After recovery WebUI does not load values

rootzoll, 2022-05-19:
On the RC1 I can go thru the whoe recovery/update process with the WebUI, but once its rebooted and the Dashboard loads after wallet unlock  I only see spinning circles that never go away:

[image]

@fusion44 this is something to look into together in Oslo.

openoms, 2022-05-23:
can confirm this on latest and same in the App screen.

rootzoll, 2022-05-30:
@fusion44 & @cstenglein ... as far as I can see on loading the dashboard checking the network I see just pings coming back as events and `locked: true` even tho the lnd is not locked (when checking thru SSH) and also the WebUI not asking to unlock ... could that be the issue?

[image]

rootzoll, 2022-05-31:
This also happens on restarts of the blitz. Then wallet is locked - WebUI askes to unlock, unlock request returns true - but no events happen after that on the subscribe SSE channel.

@fusion44 it seems this is related to the time the API is started - if I restart the systemd API service manually the next login to the WebUI is working.

rootzoll, 2022-06-14:
OK if https://github.com/fusion44/blitz_api/pull/113 gets merged we should be OK. Tests look good.

rootzoll, 2022-06-14:
Hmm Test with fresh setup of core lightning has still a problem:

[image]

So there is the message `{"bitcoin": "done", "bitcoin_msg": "", "lightning": "offline", "lightning_msg": "Unable to connect to CLN daemon, waiting..."}` but even cln is ready on the SSH side - the API is not signaling a ready state so UI keeps on waiting forever.

- calling `lightning-cli getinfo` on SSH works
- A reload of the webUI does not change anything.
- A start/stop of the blitzapi service does not change anything.

rootzoll, 2022-06-14:
Same is also happening with a fresh just Bitcoin setup. This is the message on system_startup_info `{"bitcoin": "done", "bitcoin_msg": "", "lightning": "disabled", "lightning_msg": ""}`

rootzoll, 2022-06-17:
OK fixed & tested everything so far - there is still an issue core lightnig when wallet lock is activated #3167 - but because that is not a default, its not a showblocker anymore.

rootzoll, 2022-06-20:
So there is a show stopper with the API: https://github.com/fusion44/blitz_api/issues/120

rootzoll, 2022-06-20:
OK got it fixed in the raspiblitz repo - will make a RC2 update soon for testing.

rootzoll, 2022-07-27:
This one looks good now - closing for v1.8.0 release.

closed 2022-07-27 by rootzoll