raspiblitz/raspiblitz #3140: c-lightning unexpectedly stopped working

Gilbog, 2022-05-22:
I'm on Raspiblitz 1.7.2 and c-lightning 0.10.2 and have 6 or 7 channels open. 

This morning **lightnind** had stopped working, but I had not made any changes since last night, when all still ran normally.
[image]
Bitcoin core 0.22 works normally. 

`sudo systemctl start lightningd` (as well as `restart`) or a reboot didn't fix it. Status is "loaded" and "enabled", but not "running". 

`sudo journalctl -t lightningd` is empty.

`sudo journalctl -fu lightningd -n 30` says:
-- Journal begins at Sat 2022-05-21 19:30:22 CEST. --
May 22 10:31:30 raspberrypi systemd[1]: lightningd.service: Main process exited, code=exited, status=137/n/a
May 22 10:31:30 raspberrypi systemd[1]: lightningd.service: Failed with result 'exit-code'.
May 22 10:31:30 raspberrypi systemd[1]: lightningd.service: Consumed 5.233s CPU time.
May 22 10:32:00 raspberrypi systemd[1]: lightningd.service: Scheduled restart job, restart counter is at 119.
May 22 10:32:00 raspberrypi systemd[1]: Stopped c-lightning daemon on mainnet.
May 22 10:32:00 raspberrypi systemd[1]: lightningd.service: Consumed 5.233s CPU time.
May 22 10:32:00 raspberrypi systemd[1]: Starting c-lightning daemon on mainnet...
May 22 10:32:00 raspberrypi systemd[1]: Started c-lightning daemon on mainnet.
May 22 10:32:06 raspberrypi sh[558052]: Killed
May 22 10:32:06 raspberrypi systemd[1]: lightningd.service: Main process exited, code=exited, status=137/n/a
May 22 10:32:06 raspberrypi systemd[1]: lightningd.service: Failed with result 'exit-code'.
May 22 10:32:06 raspberrypi systemd[1]: lightningd.service: Consumed 5.298s CPU time.
May 22 10:32:36 raspberrypi systemd[1]: lightningd.service: Scheduled restart job, restart counter is at 120.
May 22 10:32:36 raspberrypi systemd[1]: Stopped c-lightning daemon on mainnet.
May 22 10:32:36 raspberrypi systemd[1]: lightningd.service: Consumed 5.298s CPU time.
May 22 10:32:36 raspberrypi systemd[1]: Starting c-lightning daemon on mainnet...
May 22 10:32:36 raspberrypi systemd[1]: Started c-lightning daemon on mainnet.
May 22 10:32:42 raspberrypi sh[561913]: Killed
May 22 10:32:42 raspberrypi systemd[1]: lightningd.service: Main process exited, code=exited, status=137/n/a
May 22 10:32:42 raspberrypi systemd[1]: lightningd.service: Failed with result 'exit-code'.
May 22 10:32:42 raspberrypi systemd[1]: lightningd.service: Consumed 5.380s CPU time.
May 22 10:33:12 raspberrypi systemd[1]: lightningd.service: Scheduled restart job, restart counter is at 121.
May 22 10:33:12 raspberrypi systemd[1]: Stopped c-lightning daemon on mainnet.
May 22 10:33:12 raspberrypi systemd[1]: lightningd.service: Consumed 5.380s CPU time.
May 22 10:33:12 raspberrypi systemd[1]: Starting c-lightning daemon on mainnet...
May 22 10:33:12 raspberrypi systemd[1]: Started c-lightning daemon on mainnet.
May 22 10:33:18 raspberrypi sh[565551]: Killed
May 22 10:33:18 raspberrypi systemd[1]: lightningd.service: Main process exited, code=exited, status=137/n/a
May 22 10:33:18 raspberrypi systemd[1]: lightningd.service: Failed with result 'exit-code'.
May 22 10:33:18 raspberrypi systemd[1]: lightningd.service: Consumed 5.463s CPU time.

This may be related to my earlier issue #3104, but I had not made any changes to configuration files since that issue was fixed.

Really confused what happend ... please help, thanks!

openoms, 2022-05-23:
The CLN logs are in the SYSTEM menu or use the alias:
`cllog`

`alias cllog='sudo tail -n 30 -f /home/bitcoin/.lightning/bitcoin/cl.log'`

and check your config file for typos

Gilbog, 2022-05-23:
I think this is a problem I may have caused in my attempt to self-install a backup (see my earlier issue #3104). I thought I had fixed that, but apparently not. 

Being a newbie, I'm not sure which config file to fix, please advise.

Here is what I get from cl.log:

`admin:~ ₿ sudo tail -n 30 -f /home/bitcoin/.lightning/bitcoin/cl.log`
2022-05-23T07:49:46.142Z INFO    plugin-backup.py: root        : Comparing backup version 299611 versus first write version 299773
2022-05-23T07:49:46.142Z INFO    plugin-backup.py: Comparing backup version 299611 versus first write version 299773
2022-05-23T07:49:46.142Z INFO    plugin-backup.py: Backup is out of date, we cannot continue safely. Emergency shutdown.
2022-05-23T07:49:47.146Z INFO    plugin-backup.py: Killing process lightningd (2208327)

2022-05-23T07:50:23.066Z INFO    plugin-backup.py: root        : Comparing backup version 299611 versus first write version 299773
2022-05-23T07:50:23.067Z INFO    plugin-backup.py: Comparing backup version 299611 versus first write version 299773
2022-05-23T07:50:23.067Z INFO    plugin-backup.py: Backup is out of date, we cannot continue safely. Emergency shutdown.

2022-05-23T07:50:59.130Z INFO    plugin-backup.py: root        : Comparing backup version 299611 versus first write version 299773
2022-05-23T07:50:59.130Z INFO    plugin-backup.py: Comparing backup version 299611 versus first write version 299773
2022-05-23T07:50:59.130Z INFO    plugin-backup.py: Backup is out of date, we cannot continue safely. Emergency shutdown.
2022-05-23T07:51:00.135Z INFO    plugin-backup.py: Killing process lightningd (2216951)

2022-05-23T07:51:35.384Z INFO    plugin-backup.py: root        : Comparing backup version 299611 versus first write version 299773
2022-05-23T07:51:35.385Z INFO    plugin-backup.py: Comparing backup version 299611 versus first write version 299773
2022-05-23T07:51:35.385Z INFO    plugin-backup.py: Backup is out of date, we cannot continue safely. Emergency shutdown.

openoms, 2022-05-23:
your config file is in the SYSTEM menu

use:
`config.scripts/cl-plugin.backup.sh on`

to reinititiaize the backup plugin.

Gilbog, 2022-05-23:
Thank you! That did the trick, lightningd is up and running again. 

If it's not too much trouble, what exactly did I mess up? I had messed with the backup plugin (see #3104), but I thought I had fixed that with your previous help. 

I hadn't touched anything right before lightningd stopped working this time.

openoms, 2022-05-23:
not sure of the events, but have put an automatic check for this problem for the new release.

Gilbog, 2022-05-23:
Thanks again for your help!

closed 2022-05-23 by Gilbog