raspiblitz/raspiblitz #3142: Debian sources.list file with tor domains

barbastrejo7, 2022-05-25:
It would be nice to have the tor domains to update and install new software, here's an example:

In particular, once you have the [apt-transport-tor](https://packages.debian.org/apt-transport-tor) package installed, the following entries should work in your sources list for a Debian system:

deb  tor+http://2s4yqjx5ul6okpp3f2gaunr2syex5jgbfpfvhxxbbjwnrsvbk5v3qbid.onion/debian          buster            main
deb  tor+http://2s4yqjx5ul6okpp3f2gaunr2syex5jgbfpfvhxxbbjwnrsvbk5v3qbid.onion/debian          buster-updates    main
deb  tor+http://5ajw6aqf3ep7sijnscdzw77t7xq4xjpsy335yb2wiwgouo7yfxtjlmid.onion/debian-security buster/updates    main

openoms, 2022-05-26:
These are already added in https://github.com/rootzoll/raspiblitz/blob/c07b860c6b0bbbe253a2183847404eb4e92f8f52/home.admin/config.scripts/tor.install.sh#L152

barbastrejo7, 2022-05-29:
Yes, but that is for the tor repository, not the debian repo, right?

rootzoll, 2022-06-28:
@openoms can this issue be closed?

openoms, 2022-06-28:
The tor repo for the Debian packages is not added yet, but I am not sure if I'd want it anyway as it is significantly slower and generic packages don't give away that much.