raspiblitz/raspiblitz #3145: WebUI cannot login - 1.8.0rc1

GoetzGoerisch, 2022-05-28:
With `1.8.0rc1` image the login to the WebUI is not possible.

**Steps:**
- Upgrade running 1.7.2 installation
- Insert new SD Card
- Run through recovery/upgrade process

**Frontend error:** `An error occured: Password is wrong` (Altough PW is correct)

**Logs**:
```
May 28 13:24:20 raspberrypi systemd[1]: Started BlitzBackendAPI.
May 28 13:24:24 raspberrypi python[4746]: INFO:     Started server process [4746]
May 28 13:24:24 raspberrypi python[4746]: INFO:     Waiting for application startup.
May 28 13:24:24 raspberrypi python[4746]: WARNING:root:Setup not finished. Defering handler startup. Current phase: 'starting'
May 28 13:24:24 raspberrypi python[4746]: INFO:     Application startup complete.
May 28 13:24:24 raspberrypi python[4746]: INFO:uvicorn.error:Application startup complete.
May 28 13:24:24 raspberrypi python[4746]: INFO:     Uvicorn running on http://0.0.0.0:11111 (Press CTRL+C to quit)
May 28 13:24:24 raspberrypi python[4746]: INFO:uvicorn.error:Uvicorn running on http://0.0.0.0:11111 (Press CTRL+C to quit)
May 28 13:24:41 raspberrypi python[4746]: INFO:     127.0.0.1:34182 - "POST /api/v1/system/login HTTP/1.0" 401 Unauthorized
May 28 13:26:10 raspberrypi python[4746]: INFO:     127.0.0.1:34184 - "POST /api/v1/system/login HTTP/1.0" 401 Unauthorized
```

**Browser console logs**:
```
xhr.js:220          POST http://192.168.178.8/api/v1/system/login 401 (Unauthorized)
(anonymous) @ xhr.js:220
e.exports @ xhr.js:16
e.exports @ dispatchRequest.js:56
Promise.then (async)
f.request @ Axios.js:89
(anonymous) @ Axios.js:144
(anonymous) @ bind.js:9
(anonymous) @ Login.tsx:48
c @ runtime.js:63
(anonymous) @ runtime.js:294
(anonymous) @ runtime.js:119
r @ asyncToGenerator.js:3
l @ asyncToGenerator.js:25
(anonymous) @ asyncToGenerator.js:32
(anonymous) @ asyncToGenerator.js:21
(anonymous) @ Login.tsx:41
De @ react-dom.production.min.js:54
Ue @ react-dom.production.min.js:54
(anonymous) @ react-dom.production.min.js:55
Ar @ react-dom.production.min.js:105
Br @ react-dom.production.min.js:106
(anonymous) @ react-dom.production.min.js:117
su @ react-dom.production.min.js:273
je @ react-dom.production.min.js:52
Hr @ react-dom.production.min.js:109
Kt @ react-dom.production.min.js:74
Wt @ react-dom.production.min.js:73
Login.tsx:63 {err: a}err: acode: "ERR_BAD_REQUEST"config: {transitional: {…}, transformRequest: Array(1), transformResponse: Array(1), timeout: 0, adapter: ƒ, …}message: "Request failed with status code 401"name: "AxiosError"request: XMLHttpRequest {onreadystatechange: null, readyState: 4, timeout: 0, withCredentials: false, upload: XMLHttpRequestUpload, …}response: {data: {…}, status: 401, statusText: 'Unauthorized', headers: {…}, config: {…}, …}[[Prototype]]: Error[[Prototype]]: Objectconstructor: ƒ Object()hasOwnProperty: ƒ hasOwnProperty()isPrototypeOf: ƒ isPrototypeOf()propertyIsEnumerable: ƒ propertyIsEnumerable()toLocaleString: ƒ toLocaleString()toString: ƒ toString()valueOf: ƒ valueOf()__defineGetter__: ƒ __defineGetter__()__defineSetter__: ƒ __defineSetter__()__lookupGetter__: ƒ __lookupGetter__()__lookupSetter__: ƒ __lookupSetter__()__proto__: (...)get __proto__: ƒ __proto__()set __proto__: ƒ __proto__()
```
**Debug Log**:
https://l.termbin.com/wjipk

GoetzGoerisch, 2022-05-28:
After the inital try, I updated the OS and patch to the latest `dev` branch commit.

This was updated:
```
From https://github.com/fusion44/blitz_api
HEAD is now at bb6d6a6 chore: dep update; update version to 0.5.0
...
# BRANCH ---> main
# old commit -> a34dd7954cfc68e86c746e571b13811b5913374f
# new commit -> bb6d6a6bff3e47ba2408340c15f799e18e1e9e04
# blitzapi updates and restarted
```

```
From https://github.com/cstenglein/raspiblitz-web
HEAD is now at 5c5addc Merge pull request #304 from cstenglein/reenable-ln-dialog
...
# BRANCH ---> master
# old commit -> 8516a6a1b321cf8aed7d6f3c125a61350374c794
# new commit -> 5c5addc82ac4f6b38c216e1db82ae65390894a2d
# reload WebUI in your browser
```

Latest debug log: https://l.termbin.com/4ovs

rootzoll, 2022-05-31:
Can you try on the ssh terminal what the following script returns:
```
/home/admin/config.scripts/blitz.passwords.sh check a "[YOURPASSWORD]"
```

GoetzGoerisch, 2022-05-31:
```
runningUser='0'
# salt file exists
# first try within last minute - respond fast
correct=1
```

fusion44, 2022-06-30:
I've never seen such behaviour until now. The error you see is just unauthorized which is normally a invalid auth token.

rootzoll, 2022-06-30:
OK that error seem to come from the password checking shellscript - that due to moving API from root to blitzapi needed some sudos to run also as that new user. I adpated the shellscript and that strange error is gone. BUT I think that was not the original problem of this issue.

rootzoll, 2022-07-27:
Closing this issue for now because there were no further reports on it - to be reopened if shows up again.

closed 2022-07-27 by rootzoll

psilonux, 2023-04-09:
I'm having trouble to login on the webgui.
`Apr 09 21:05:05 raspberrypi python[506114]: E0409 21:05:05.153078190  506114 fork_posix.cc:76]           Other threads are currently calling into gRPC, skipping fork() handlers
Apr 09 21:05:05 raspberrypi sudo[528065]: blitzapi : PWD=/home/blitzapi/blitz_api ; USER=root ; COMMAND=/usr/bin/ls /mnt/hdd/app-data/passwords/salt.txt
Apr 09 21:05:05 raspberrypi sudo[528065]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=1003)
Apr 09 21:05:05 raspberrypi sudo[528065]: pam_unix(sudo:session): session closed for user root
Apr 09 21:05:05 raspberrypi sudo[528068]: blitzapi : PWD=/home/blitzapi/blitz_api ; USER=root ; COMMAND=/usr/bin/cat /mnt/hdd/app-data/passwords/salt.txt
Apr 09 21:05:05 raspberrypi sudo[528068]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=1003)
Apr 09 21:05:05 raspberrypi sudo[528068]: pam_unix(sudo:session): session closed for user root
Apr 09 21:05:06 raspberrypi sudo[528170]: blitzapi : PWD=/home/blitzapi/blitz_api ; USER=root ; COMMAND=/usr/bin/cat /mnt/hdd/app-data/passwords/a.hash
Apr 09 21:05:06 raspberrypi sudo[528170]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=1003)
Apr 09 21:05:06 raspberrypi sudo[528170]: pam_unix(sudo:session): session closed for user root
Apr 09 21:05:06 raspberrypi python[506114]: INFO:     127.0.0.1:52844 - "POST /system/login HTTP/1.0" 200 OK
Apr 09 21:05:06 raspberrypi python[506114]: INFO:     127.0.0.1:52854 - "GET /sse/subscribe HTTP/1.0" 403 Forbidden
Apr 09 21:05:06 raspberrypi python[506114]: INFO:     127.0.0.1:52860 - "GET /lightning/list-all-tx?reversed=true HTTP/1.0" 200 OK`

psilonux, 2023-04-09:
I'm having trouble logging into the WebUI (v1.8):

> Apr 09 21:05:05 raspberrypi python[506114]: E0409 21:05:05.153078190  506114 fork_posix.cc:76]           Other threads are currently calling into gRPC, skipping fork() handlers
Apr 09 21:05:05 raspberrypi sudo[528065]: blitzapi : PWD=/home/blitzapi/blitz_api ; USER=root ; COMMAND=/usr/bin/ls /mnt/hdd/app-data/passwords/salt.txt
Apr 09 21:05:05 raspberrypi sudo[528065]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=1003)
Apr 09 21:05:05 raspberrypi sudo[528065]: pam_unix(sudo:session): session closed for user root
Apr 09 21:05:05 raspberrypi sudo[528068]: blitzapi : PWD=/home/blitzapi/blitz_api ; USER=root ; COMMAND=/usr/bin/cat /mnt/hdd/app-data/passwords/salt.txt
Apr 09 21:05:05 raspberrypi sudo[528068]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=1003)
Apr 09 21:05:05 raspberrypi sudo[528068]: pam_unix(sudo:session): session closed for user root
Apr 09 21:05:06 raspberrypi sudo[528170]: blitzapi : PWD=/home/blitzapi/blitz_api ; USER=root ; COMMAND=/usr/bin/cat /mnt/hdd/app-data/passwords/a.hash
Apr 09 21:05:06 raspberrypi sudo[528170]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=1003)
Apr 09 21:05:06 raspberrypi sudo[528170]: pam_unix(sudo:session): session closed for user root
Apr 09 21:05:06 raspberrypi python[506114]: INFO:     127.0.0.1:52844 - "POST /system/login HTTP/1.0" 200 OK
Apr 09 21:05:06 raspberrypi python[506114]: INFO:     127.0.0.1:52854 - "GET /sse/subscribe HTTP/1.0" 403 Forbidden
Apr 09 21:05:06 raspberrypi python[506114]: INFO:     127.0.0.1:52860 - "GET /lightning/list-all-tx?reversed=true HTTP/1.0" 200 OK