raspiblitz/raspiblitz #3147: Peer count low during node sync in version 1.7.2

EFiend, 2022-05-29:
Hi,

I am running version 1.7.2 on an RPI 4 8gb and my peer count during the node syncing phase is low. Typically I will have less than 3 peers and often I have 1 or zero peers. The node is connected to the Tor ( by default ). I initially tried the self-sync to build the initial blockchain but after a month the progress was around ~60%. I decided then to download the blockchain via Bitcoin Core on a separate computer, and was able to download it in less than 2 days with typically 10 peers connected to Bitcoin Core. I re-flashed the RPI microSD and wiped the SSD attached to the RPI, and scp file transferred the blockchain over. Now the Blockchain progress is at 99.9% but still am only connected to 3 peers. When I run `bitcoin-cli getnetworkinfo | grep "connections"` , I get:

```
  "connections": 3,
  "connections_in": 1,
  "connections_out": 2,
```
Prior to me scp file transferring the blockchain over, I tried manually adding nodes from bitnode.io but either a) they did not connect or b) would not stay connected for long. Additionally, prior to this most recent microSD re-flash, I would restart the RPI to see if that would help in increasing the peers but it did not ( would stay with typically less than 3 peers ). I have attached the debug.log for reference below. 

Any help would be greatly appreciated in increasing peer count.

______________________________________________________________________________________________________________________________________________

***************************************************************
* RASPIBLITZ DEBUG LOGS
***************************************************************
blitzversion: 1.7.2
chainnetwork: bitcoin / main

***************************************************************
* RASPIBLITZ DEBUG LOGS
***************************************************************
blitzversion: 1.7.2
chainnetwork: bitcoin / main
 01:26:23 up 37 min,  2 users,  load average: 2.08, 2.04, 2.01

*** SETUPPHASE / BOOTSTRAP ***
see logs: cat /home/admin/raspiblitz.log
setupPhase--> done
state--> ready

*** BACKGROUNDSERVICE ***
to monitor Background service call: sudo journalctl -f -u background

*** BLOCKCHAIN (MAINNET) SYSTEMD STATUS ***
● bitcoind.service - Bitcoin daemon
     Loaded: loaded (/etc/systemd/system/bitcoind.service; enabled; vendor preset: enabled)
     Active: active (running) since Sun 2022-05-29 01:01:42 BST; 24min ago
    Process: 5321 ExecStartPre=/home/admin/config.scripts/blitz.systemd.sh log blockchain STARTED (code=exited, status=0/SUCCESS)
    Process: 5330 ExecStartPre=/bin/chgrp bitcoin /mnt/hdd/bitcoin (code=exited, status=0/SUCCESS)
    Process: 5331 ExecStart=/usr/local/bin/bitcoind -daemonwait -pid=/mnt/hdd/bitcoin/bitcoind.pid -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin -debuglogfile=/mnt/hdd/bitcoin/debug.log (code=exited, status=0/SUCCESS)
   Main PID: 5423 (bitcoind)
      Tasks: 18 (limit: 2198)
        CPU: 4min 38.767s
     CGroup: /system.slice/bitcoind.service
             └─5423 /usr/local/bin/bitcoind -daemonwait -pid=/mnt/hdd/bitcoin/bitcoind.pid -conf=/mnt/hdd/bitcoin/bitcoin.conf -datadir=/mnt/hdd/bitcoin -debugl…

May 29 00:50:10 raspberrypi systemd[1]: Starting Bitcoin daemon...
May 29 01:01:42 raspberrypi systemd[1]: Started Bitcoin daemon.

*** LAST BLOCKCHAIN (MAINNET) ERROR LOGS ***
sudo journalctl -u bitcoind -b --no-pager -n8
-- Journal begins at Fri 2022-01-28 03:21:52 GMT, ends at Sun 2022-05-29 01:26:24 BST. --
May 29 00:50:10 raspberrypi systemd[1]: Starting Bitcoin daemon...
May 29 01:01:42 raspberrypi systemd[1]: Started Bitcoin daemon.

*** LAST BLOCKCHAIN (MAINNET) 20 INFO LOGS ***
sudo tail -n 20 /mnt/hdd/bitcoin/debug.log
2022-05-29T00:01:42Z init message: Starting network threads…
2022-05-29T00:01:42Z DNS seeding disabled
2022-05-29T00:01:42Z net thread start
2022-05-29T00:01:42Z addcon thread start
2022-05-29T00:01:42Z opencon thread start
2022-05-29T00:01:42Z init message: Done loading
2022-05-29T00:01:42Z msghand thread start
2022-05-29T00:01:54Z Leaving InitialBlockDownload (latching to false)
2022-05-29T00:01:58Z New outbound peer connected: version: 70016, blocks=738340, peer=0 (block-relay-only)
2022-05-29T00:02:02Z New outbound peer connected: version: 70016, blocks=738340, peer=1 (block-relay-only)
[... 208 lines cut ...]
btc_default_synced="0"
btc_default_port="8333"
btc_default_blocks_verified="738311"
btc_default_blocks_headers="738340"
btc_default_online="1"
btc_default_error_full=""
btc_default_error_short=""
btc_default_version="v22.0.0"
btc_default_sync_progress="0.9999307045988485"
btc_default_sync_percentage="99.99"
btc_default_sync_initial_done="0"
btc_default_ready="1"
btc_default_running="1"
btc_default_sync_initialblockdownload="0"
btc_default_activated="1"
btc_default_blocks_behind="29"

*** OPTION: SHARE THIS DEBUG OUTPUT ***
An easy way to share this debug output on GitHub or on a support chat
Use the following command and share the resulting link using termbin.com service and tor proxy:
 debug -l
If tor is failing and you don't mind leaking your ip address to the termbin service, use without tor:
 debug -l -n

Press ENTER to return to main menu.

EFiend, 2022-05-29:
Update:

Node has finished syncing and has 10 peers on the mainnet and 3 peers on LND.  I'm not sure if I will have to start over again in the future with the selfsync, but if possible it would be nice to know of methods to increase the peers during self-sync. 

Let me know if I should close this issue, or if there are any follow-up questions or comments.

EFiend, 2022-05-30:
Thank- you for the follow up. I had read that there was the kickstart it if detects 0 peers, but when there would be under 3 peers I sometimes wondered if the blockchain could be corrupted in the initial download. If the small number of nodes you are connected to could send you an incorrect blockchain and if your system could only rely on this small number of nodes to check consensus I wondered if a node could finish syncing an incorrect blockchain and then enter the network with an incorrect ledger. I, however, am a novice to nodes so I do not know if this is possible or not. 

The node is working well currently so I will close this issue.

rootzoll, 2022-05-31:
I will keep this ticket open as a reminer to look into for v1.8.2

closed 2023-06-07 by cstenglein

scubafly, 2023-07-26:
I had a similar issue in 1.9.0 Even with 0 peers.

When I run:

`bitcoin-cli getnetworkinfo | grep "connections"`

I get:

 ```
"connections": 1,
  "connections_in": 0,
  "connections_out": 1,
```

Running `/home/admin/config.scripts/bitcoin.monitor.sh mainnet peer-kickstart auto` gives me a few peers, and that in drops again to 0

cstenglein, 2023-07-26:
@scubafly 
It should help if you restart tor, since that is usually the issue.

scubafly, 2023-07-26:
> @scubafly 
> 
> It should help if you restart tor, since that is usually the issue.

Looks like that did the trick. Now I have 11 peers :)