raspiblitz/raspiblitz #3149: LND RPC services not available

Yonben, 2022-05-29:
Hi,
I changed my node alias and since I did, LND was stuck at "prepare sync".
I tried many things.
The debug log looked like that:
https://l.termbin.com/enj46

Now after a few hours, I decided to open an issue. The `debug -l` said "unable to rename" (exactly like https://github.com/rootzoll/raspiblitz/issues/2996).
I "sudo rm /var/cache/raspiblitz/debug.log" to be able to `debug -l`. As you can see with the link I previously shared it worked but even weirder... LND just started to work?!

My issue seems fixed but I wanted to raise awareness, maybe something is worth investigating there? Maybe it's just coincidental?

Thanks!

rootzoll, 2022-05-30:
Was your name maybe to long (max should be 32 chars)? 

But in general ... try to rename it back to the old name by using CTRL+C to exit to terminal and then call command `menu` > SYSTEM > LNDCONF .. save & restart LND - best also to restart RaspiBlitz with 'restart'

Yonben, 2022-05-31:
As I mentioned, it just worked by itself suddenly...
However I will use restart and not reboot from now on. Thanks!

closed 2022-06-01 by rootzoll