raspiblitz/raspiblitz #3152: Have channel opens unnanounced by default

Bosch-0, 2022-05-31:
Raspiblitz should default to new channel opens being unannounced. Public channels should be opt in as they are primarily for routing node operators which the majority of lightning users are not. 

When opening channels unannounced as default it would be handy to have a tool tip that pointed users who do want to be a routing node towards enabling a public channel open. Users who are looking to be routers would probably know the difference between public and unannounced though but this is a little extra life improvement. This is important because a channel needs to be closed and re-opened to change it's public status. 

### What benefits does this have?

- More reliable network wide payments as you wouldn't have inefficiently allocated capital acting as de-facto routing nodes (I'd assume many Raspiblitz users are opening nodes to make payments and doing zero channel management).  These channels also generally have low fees (the default) so get selected for routes more often than not which as the liquidity is not managed results in more payment failures for all users on lightning. 

- Users who open channels to just make payments (I'd assume most) wouldn't have to deal with all the UX implications of routing payments. 

- Some added privacy for users, they can still be revealed through probing but it makes it that little bit harder. 

- Some marginal reduction in gossip on the network limiting bandwidth requirements for node operators. This is likely going to become problematic with onion messages coming to lightning so limiting this as much as possible is a minor + to me. 

We recently implemented this in Zeus: https://github.com/ZeusLN/zeus/pull/1002

openoms, 2022-05-31:
Mostly agree, but noting that we only have power to change the SSH menu (and maybe the WebUI) defaults, but most users are (anecdotally) using interfaces like RTL, Thunderhub and of course Zeus to open channels.
Also this is somewhat similar to the default fee question discussed here: https://github.com/rootzoll/raspiblitz/issues/1980
I don't think that any of these tools are meant to be too opinionated and should rely mostly on the underlying config option for LND / CLN. Maybe it would be the best place to configure unannounced channel by default on the lnd.conf / CLN config file - but so far that is option is not available.

openoms, 2023-07-25:
There is no action to be taken for the CLI. Also on an always-on node people probably expect to be able to route on their channels by default. Closing for now.

closed 2023-07-25 by openoms