raspiblitz/raspiblitz #3157: [feature request] ability to disable `lnd.check.sh` for guided hybrid mode setup

blckbx, 2022-06-04:
We're planning to provide a guided, pre-configured clearnet over VPN solution that utilizes a setup script for general node configuration. Based on the underlying node software used the scripts decides which paths to take and parameters to set. 

Specifically we want to adjust `lnd.conf` to our needs which in detail means to
- check for and remove/uncomment existing entries for `externalip`, `externalhosts` and replace with our own entry
- check for and (if necessary) add `listen=0.0.0.0:9735`
- check for and adjust `tor.streamisolation` to `false`
- check for and adjust `tor.skip-proxy-for-clearnet-targets` to `true`

I think we would interfere with RaspiBlitz' built-in `lnd.check.sh` watch script. Any hint appreciated on what we could do to provide the best of both worlds.

openoms, 2022-06-04:
To disable the `lnd.check.sh` entirely could just delete or rename it.
It is called in the `lnd.service` buy not necessary to run.

blckbx, 2022-06-04:
> It is called in the `lnd.service` but not necessary to run.

Thanks for your reply!

If systemd service keeps starting without `lnd.check.sh`, that would be a good and simple solution for our use case. 
So we can either remove `lnd.check.sh` or comment out `ExecStartPre` in `lnd.service`.

openoms, 2022-06-04:
Yes, the `-` sign after `ExecStartPre=` means that there will be no error if the file is not there.

blckbx, 2022-06-04:
Oh interesting! I didn't know that. Thanks a lot!

openoms, 2022-06-04:
Looks like you will have a solution to this issue https://github.com/rootzoll/raspiblitz/issues/2855

openoms, 2022-06-06:
related: https://github.com/rootzoll/raspiblitz/issues/3094

`tor.streamisolation` won't be switched automatically