raspiblitz/raspiblitz #3158: Can't get to the menu from status screen

Nerfzooka, 2022-06-04:
My headless raspiblitz is running well, maybe too well, it won't exit from the status screen when I ssh in.  The FAQ says "SSH into your RaspiBlitz and EXIT to terminal from the menu."  I tried EXIT, exit, control x, control d, escape, control q, \q, quit and whatever else I could think of, but only control c ended the status screen by cutting off the ssh connection.  How do you get to the menu while sshing?

openoms, 2022-06-04:
Use the `admin` user to log in, not `pi`

Nerfzooka, 2022-06-04:
oooooh, of course.   Thanks openoms
[image]
!

closed 2022-06-06 by rootzoll