raspiblitz/raspiblitz #3159: Check Umbrel migration for new 0.5.0 version

rootzoll, 2022-06-08:
https://blog.getumbrel.com/introducing-umbrel-0-5-a-beautiful-personal-server-os-for-self-hosting-dcbee7e23b64

Also now that they have also core lightning - this needs to be part of the migration process, too.

openoms, 2022-06-09:
This helps moving to a docker based setup, but just for FYI: https://github.com/dsbaars/unbrel

rootzoll, 2022-06-09:
unbrel is a nice hack of Umbrel ;) ... but the topic of this issue is more to adapt to their changes HDD layout (the moved lots of files around - but I already have that adapted) and to support also migratiing the new Core Lightning data to raspiblitz. See progress in branch: https://github.com/rootzoll/raspiblitz/tree/3159umbrel

rootzoll, 2022-06-09:
OK implemented - also the core lightning migration from raspiblitz 0.5.0 to raspiblitz 1.8.0
final testing during RC2

rootzoll, 2022-07-27:
No bad reports during rc phase - closing issue for v1.8.0 release.

PR #3160 (2022-06-09): Adapt to new migration for umbrel 0.5.0 including core lightning data
fixes #3159
closed 2022-07-27 by rootzoll