raspiblitz/raspiblitz #3162: No wallet and no macaroons - Debian 11.2 - VM on Proxmox

mark0st83, 2022-06-10:
Seems like a problem with pip.

Same with bench dev (v1.8R1) and v1.7

After a `build_sdcard.sh` with a fresh Debian11 the setup stucks in „LND Credentials“ and after the 2 min the setup stopped:

```
Calling _provision.setup.sh for basic setup tasks ..
Follow in a new terminal with: 'tail -f raspiblitz.provision-setup.log'
***********************************************************
ERROR - please report to development team
***********************************************************
state(error) message(lnd-no-macaroons)
https://github.com/rootzoll/raspiblitz#support
command to shutdown --> off
```

Try to make new macaroons over the menu failed. (Found no files)
`lnd.initwallet.py`:
```
admin@10.0.0.113:~ ₿ sudo /home/admin/config.scripts/lnd.initwallet.py new mainnet password
Traceback (most recent call last):
  File "/home/admin/config.scripts/lnd.initwallet.py", line 31, in <module>
    from lndlibs import walletunlocker_pb2 as lnrpc
  File "/home/admin/config.scripts/lndlibs/walletunlocker_pb2.py", line 14, in <module>
    from . import lightning_pb2 as lightning__pb2
  File "/home/admin/config.scripts/lndlibs/lightning_pb2.py", line 33, in <module>
    _descriptor.EnumValueDescriptor(
  File "/usr/local/lib/python3.9/dist-packages/google/protobuf/descriptor.py", line 755, in __new__
    _message.Message._CheckCalledFromGeneratedFile()
TypeError: Descriptors cannot not be created directly.
If this call came from a _pb2.py file, your generated code is out of date and must be regenerated with protoc >= 3.19.0.
If you cannot immediately regenerate your protos, some other possible workarounds are:
 1. Downgrade the protobuf package to 3.20.x or lower.
 2. Set PROTOCOL_BUFFERS_PYTHON_IMPLEMENTATION=python (but this will use pure-Python parsing and will be much slower).

More information: https://developers.google.com/protocol-buffers/docs/news/2022-05-06#python-updates
```

I found a possible solution on GitHub [here](https://github.com/protocolbuffers/protobuf/issues/10051) and [here](https://github.com/deepmind/alphafold/issues/478)

I added `pip3 install --upgrade protobuf==3.20.0` in the build_sdcard.sh on line 307

`python_libs="grpcio==1.42.0 googleapis-common-protos==1.53.0 toml==0.10.2 j2cli==0.3.10 requests[socks]==2.21.0 protobuf==3.20.0"`

The build with Python 3.9.2 and bench v1.7 seems to work. The 1.8r1 seems to work with a empty SSD, but it looks like it stucks with delete data and use blockchain.
But maybe this is just a other problem.

To switch on Python 3.10.1 like in the guide seems to work aswell.

rootzoll, 2022-06-13:
@openoms are there some protobuffs we need to update for v1.8.0?

openoms, 2022-06-13:
It seems to need protobuff > 3.19.0.
I have 3.19.1 on my amd64 server based on Debian 11 and on my test RPi. That should work.

what was your original protobuff version @mark0st83?

openoms, 2022-06-13:
Thank you for the report, it is indeed a pip mess. 
Running:
```
sudo python3 -m pip install grpcio-tools
```

Installed protobuff protobuf-3.20.1:
```
Collecting protobuf<4.0dev,>=3.12.0
  Downloading protobuf-3.20.1-cp39-cp39-manylinux_2_5_x86_64.manylinux1_x86_64.whl (1.0 MB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 1.0/1.0 MB 5.4 MB/s eta 0:00:00
 ```
 
 But if protobuff 3.12 is already present it does not upgrade as it meets the requirement.

Let's add the upgrade to the dependencies in build_sdcard.sh so all have the right versions.

openoms, 2022-06-13:
So protobuff 3.19.1 seems to work ok, but the latest is protobuf-3.20.1.
Shall we pin the latest?

mark0st83, 2022-06-13:
I will Setup a 1.8r1 with the latest later.

openoms, 2022-06-13:
The change did not break anything on live systems:
```
python_libs="grpcio==1.42.0 googleapis-common-protos==1.53.0 toml==0.10.2 j2cli==0.3.10 requests[socks]==2.21.0 protobuf==3.20.1"
sudo -H python3 -m pip install --upgrade pip
sudo -H python3 -m pip install ${python_libs}
```

mark0st83, 2022-06-13:
I just saw that I’ve already tested a bunch of settings.

I’ve  already tested:

V1.8r1
Python 3.9.2
Protobuf 0.20.1

V1.8r1
Python 3.9.2
Protobuf 0.20.1

now live:
V1.7
Python 3.9.2
Protobuf 0.20.0

And several settings with Python 3.10.1, but that makes no different to me.

so I’m thinking 0.20.1 would be fine for the Realeasecandidate.

rootzoll, 2022-07-27:
OK tested fine so far - closing for v1.8.0 release.

closed 2022-07-27 by rootzoll