raspiblitz/raspiblitz #3163: Unable to reactivate LND Static Channel Backup on Nextcloud

benniblitz, 2022-06-11:
I've been having a few issues reactivating LND static channel back up on Nextcloud after the latest update.

It used to work well for a few months, then it stopped working.

I use 

Through the SSH menu I activate the LND Static Channel Backup

Nextcloud Setting changed ..
rm: cannot remove '/var/cache/raspiblitz/.tmp' : Operation not permitted
sed: -e expression #1, char 59: unknown option to ‘s’
err=’Nextcloud credentials are missing’

rootzoll, 2022-06-13:
Maybe related and fixed in issue #3116 which will be part of upcoming v1.8.0 release. Please test if an update to v1.8.0 is the fixes this once its available and report back.

cstenglein, 2023-06-07:
@benniblitz  is this still relevant in 1.9.0? Closing for now, if it's still relevant, please re-open

closed 2023-06-07 by cstenglein