raspiblitz/raspiblitz #3167: BackendAPI detect core lightning locked wallet

rootzoll, 2022-06-17:
I have a CLN node with wallet locked (which is not default). After a restart the WebUI is spinning/waiting - it gets a message `{"bitcoin": "done", "bitcoin_msg": "", "lightning": "offline", "lightning_msg": "Unable to connect to CLN daemon, waiting..."}`. When I go into SSH it asks me to unlock the wallet and when I do the WebUI kicks in all good.

@fusion44 GRPC is not working before wallet unlock - maybe use on the redis value `ln_cl_mainnet_locked` to check and signal that to the WebUI?

This issue is not an RC2 blocker - but it would be good to have it for 1.8.0 final release.

rootzoll, 2022-07-06:
Just retested with rc3 - still there. @openoms & @fusion44 do you see a quick fix here or should I push to v1.8.1?

openoms, 2022-07-07:
As with everything not supported by the `cln-grpc` would need to fall back to the CLI.
if on redis `ln_cl_mainnet_locked` need to run
`config.scripts/cl.hsmtool.sh unlock`
currently it works only to take a password from the ssh dialog so will need to look into refactoring there.

rootzoll, 2022-07-07:
@fusion44 how do we fix this on the API side .. will you find time over the weekend or shoudl i take a look?

fusion44, 2022-07-07:
I don't think this is a big change. I should be able to do this over the weekend.

fusion44, 2022-07-10:
Unlock via SSH works. Unlock via API not implemented yet. What is the script I must call with the user supplied password?

I'm keeping track of the changes in https://github.com/fusion44/blitz_api/tree/blitz-cln-unlock

openoms, 2022-07-10:
@fusion44
>What is the script I must call with the user supplied password?

To unlock the CLN wallet use;
```
/home/config.scripts/cl.hsmtool.sh unlock mainnet PASSWORD_C
```
as implememted in #3211. There is also some help with errors in the exit codes.

fusion44, 2022-07-10:
Pushed my latest changes. Unlock via API should work as expected too. I'm currently on the road and can't test on a real Blitz though. Wrote my own little fake bash script to test it on my laptop.

@openoms can you try the blitz-cln-unlock branch at some point?

fusion44, 2022-07-11:
No need to test. I have some more changes to make.

fusion44, 2022-07-11:
Unlocking via SSH works but via API doesn't. 

This is how I call the script:
```bash
/home/admin/config.scripts/cl.hsmtool.sh unlock mainnet test1234
```

The script returns this error:
```
Jul 11 16:43:28 python[258027]: DEBUG:root:ProcessResult:
Jul 11 16:43:28 python[258027]: return_code: 1
Jul 11 16:43:28 python[258027]: stdout: # Cancelled
Jul 11 16:43:28 python[258027]:  
Jul 11 16:43:28 python[258027]: stderr: shred: /dev/shm/.cl.pw: failed to open for writing: No such file or directory
```
Doesn't seem like an error on the API side. Looking at the script it thinks that I've pressed cancel within the UI 
https://github.com/rootzoll/raspiblitz/blob/b33d2fcbd7b2dfca803933897836b3ccf4634771/home.admin/config.scripts/cl.hsmtool.sh#L83-L87

fusion44, 2022-07-11:
Unlocking via SSH works but via API doesn't. 

This is how I call the script:
```bash
/home/admin/config.scripts/cl.hsmtool.sh unlock mainnet password
```

The script returns this error:
```
Jul 11 16:43:28 python[258027]: DEBUG:root:ProcessResult:
Jul 11 16:43:28 python[258027]: return_code: 1
Jul 11 16:43:28 python[258027]: stdout: # Cancelled
Jul 11 16:43:28 python[258027]:  
Jul 11 16:43:28 python[258027]: stderr: shred: /dev/shm/.cl.pw: failed to open for writing: No such file or directory
```
Doesn't seem like an error on the API side. Looking at the script it thinks that I've pressed cancel within the SSH UI 
https://github.com/rootzoll/raspiblitz/blob/b33d2fcbd7b2dfca803933897836b3ccf4634771/home.admin/config.scripts/cl.hsmtool.sh#L83-L87

openoms, 2022-07-11:
the other issue could be that it doesn't work with the user `blitzapi` so rather call with with sudo.

rootzoll, 2022-07-12:
@openoms is this still a blocker for the v1.8.0 final release?

openoms, 2022-07-12:
I think so, did not test the latest blitzapi yet, but the script seems to work as intended.

fusion44, 2022-07-12:
When I call the script via SSH as shown above, the "enter password c screen" is shown in the shell. When I hit cancel the same error message as I get when called via Python is echoed.

fusion44, 2022-07-12:
Yes, thats it. RC3 included an old version of the script. I've manually replaced the code with the latest from github and everything works as expected.

PR #3211 (2022-07-07): cl.hsmtool.sh unlock: password as an option
changes to fix #3167
```
usage:
  cl.hsmtool.sh [unlock] <mainnet|testnet|signet> <password>
    success: exit 0"
    wrong password: exit 2"
    fail to unlock after 1 minute + show logs: exit 3"
```
closed 2022-07-27 by rootzoll