raspiblitz/raspiblitz #3173: [amd64] WebUI - An error occured: Password is wrong

openoms, 2022-06-20:
Installed the blitz.web.api and blitz.web.ui on my amd64 server.

Tried to log in:
`An error occured: Password is wrong`

Changed the PasswordA from the menu and restarted, but it is still the same.

@fusion44 @rootzoll can you please explain how the PasswordA is dealt with here?
Would worth a short entry in the SECURITY.md as well.

GoetzGoerisch, 2022-06-20:
relates to #3145

openoms, 2022-06-20:
```
₿ sudo /home/admin/config.scripts/blitz.passwords.sh check a "MYPASSWORD"
runningUser='0'
# salt file exists
# hashedPasswordSalt(88f44031c0badc2b)
# first try within last minute - respond fast
/home/admin/config.scripts/blitz.passwords.sh: line 76: mkpasswd: command not found
error='password cannot be checked - no hash available'
correct=0
```

so need to check for `mkpasswd`