raspiblitz/raspiblitz #3174: Unable to backup channels for C-Lightning

stn021, 2022-06-20:
Hello,

for C-Lightning there are several options for backing up the channels.
None of these work on raspiblitz.

The entry "wallet=..." in the config-file fails with "cannot open database", maybe due to user-privileges of users bitcoin and admin.
The backup-plugin does not work either, neither on raspiblitz nor on the original Core-Lightning. In https://github.com/rootzoll/raspiblitz/issues/2983 I found how to change raspiblitz/home.admin/config.scripts/cl-plugin.backup.sh. But afterwards the initialization 'config.scripts/cl-plugin.backup.sh on' failed because of No such file or directory: '/home/bitcoin/cl-plugins-available/plugins/backup/requirements.txt'

I have been at this for a while not and could not find how to backup the CLI-channels on raspiblitz.
Please let me know how I can do that.

THX, regards, stn

openoms, 2022-06-20:
The backup plugin does mirror the CLN channel database to the SDcard by default in `/home/bitcoin`. If your changes are not working now try running:
`patch` 
(same as UPDATE - PATCH)
or flash a new SDcard.

openoms, 2022-06-21:
As above, see:

`sudo ls -la /home/bitcoin/`

openoms, 2022-06-21:
Here it is:
```
₿ sudo ls -la /home/bitcoin/lightningd.sqlite3.backup
-rw-r--r-- 1 bitcoin bitcoin 208668154 Jun 21 22:16 /home/bitcoin/lightningd.sqlite3.backup
```

It is recreated (and the old one renamed with a date and saved) when you run 
```
config.scripts/cl-plugin.backup.sh on
```

Can try running the commands:
```
config.scripts/cl-plugin.backup.sh off
config.scripts/cl-plugin.backup.sh on
```

or reflash the SDcard if it does not work for you.

stn021, 2022-06-21:
The last several times I ran 'patch' it refused to run and demanded that I commit first.

This time it actually worked.
And finally the backup-file is there too.

Thank you.

openoms, 2022-06-22:
also al

closed 2022-06-22 by openoms

0xtrr, 2022-07-21:
> The last several times I ran 'patch' it refused to run and demanded that I commit first.

How did you fix this issue @stn021? I'm having the exact same problem with the plugin but when I try to run patch it just errors on 
```
admin@192.168.1.49:~ ₿ patch
# your active GitHubUser is: rootzoll
# your wanted GitHubUser is: rootzoll
# OK
# your active branch is: v1.7
# your wanted branch is: v1.7
# OK
# *** SYNCING RASPIBLITZ CODE WITH GITHUB ***
# This is for developing on your RaspiBlitz.
# THIS IS NOT THE REGULAR UPDATE MECHANISM
# and can lead to dirty state of your scripts.
# REPO ----> origin	https://github.com/rootzoll/raspiblitz.git (push)
# BRANCH --> v1.7
# ******************************************
error: cannot pull with rebase: You have unstaged changes.
error: please commit or stash them.
# COPYING from GIT-Directory to /home/admin/
# - basic admin files
# - asset directory
# - config.scripts directory
# - setup.scripts directory
# ******************************************
# Syncing Webcontent ..
# Checking if the content of BlitzPy changed ..
# checkSumBlitzPyBefore = 2e35e76c81cad5e7c909542d94940c95  -
# checkSumBlitzPyAfter  = 2e35e76c81cad5e7c909542d94940c95  -
# BlitzPy did not changed.
# ******************************************
# OK - shell scripts and assets are synced
# Reboot recommended

```

openoms, 2022-07-21:
to solve this reset you source code directory:
```
cd
rm -rf ./raspiblitz
git clone https://github.com/rootzoll/raspiblitz
patch
```