raspiblitz/raspiblitz #3178: Monitor bitcoin log for certain errors and auto fix when possible

fusion44, 2022-06-21:
I had the following Bitcoin Core error after an update to 1.8.0 RC2 from today:

```
2022-06-21T19:55:04Z HTTP: creating work queue of depth 16
2022-06-21T19:55:04Z Config options rpcuser and rpcpassword will soon be deprecated. Locally-run instances may remove rpcuser to use cookie-based auth, or may be replaced with rpcauth. Please see share/rpcauth for rpcauth auth generation.
2022-06-21T19:55:04Z HTTP: starting 4 worker threads
2022-06-21T19:55:04Z Using /16 prefix for IP bucketing
2022-06-21T19:55:04Z init message: Loading P2P addresses…
2022-06-21T19:55:05Z Error: Invalid or corrupt peers.dat (Checksum mismatch, data corrupted). If you believe this is a bug, please report it to https://github.com/bitcoin/bitcoin/issues. As a workaround, you can move the file ("/mnt/hdd/bitcoin/peers.dat") out of the way (rename, move, or delete) to have a new one created on the next start.
2022-06-21T19:55:05Z Shutdown: In progress...
2022-06-21T19:55:05Z scheduler thread exit
2022-06-21T19:55:05Z Shutdown: done
```
Deleting the file and a reboot fixed it.

Would be a nice feature to detect and fix these "small" problems via a script.

openoms, 2022-06-21:
Never seen this one, but sure could be a nice feature - especially with the most common reindexing need, but the problem is it might take days to do.

rootzoll, 2023-01-12:
Just to collect another "corrupt peers.dat" error message:
```
2023-01-06T15:14:14Z Using /16 prefix for IP bucketing
2023-01-06T15:14:14Z init message: Loading P2P addresses…
2023-01-06T15:14:14Z Error: Invalid or corrupt peers.dat (Invalid network magic number). If you believe this is a bug, please report it to https://github.com/bitcoin/bitcoin/issues. As a workaround, you can move the file ("/mnt/hdd/bitcoin/peers.dat") out of the way (rename, move, or delete) to have a new one created on the next start.
2023-01-06T15:14:14Z Shutdown: In progress...
2023-01-06T15:14:14Z scheduler thread exit
2023-01-06T15:14:14Z Shutdown: done
```

rootzoll, 2023-01-13:
Another "reindex" error message:
```
2023-01-13T00:38:53Z You can use -debug=leveldb to get more complete diagnostic messages
2023-01-13T00:38:53Z Fatal LevelDB error: Corruption: 6 missing files; e.g.: /mnt/hdd/bitcoin/chainstate/1352956.ldb
2023-01-13T00:38:53Z : Error opening block database.
Please restart with -reindex or -reindex-chainstate to recover.
2023-01-13T00:38:53Z Aborted block database rebuild. Exiting.
2023-01-13T00:38:53Z Shutdown: In progress...
2023-01-13T00:38:53Z scheduler thread exit
2023-01-13T00:38:54Z Shutdown: done
```

openoms, 2023-07-02:
just came across this after a cold restart. Agree that it could be fixed automatically with:
```
sudo -u bitcoin rm /mnt/hdd/bitcoin/peers.dat
```
if appears in the logs. Could be the the part of `bitcoind.check.sh`

rootzoll, 2023-07-29:
Check for more errors messages tagged on issues with label "Error-Analysis"