raspiblitz/raspiblitz #3179: [v1.8.0RC2] Tor package version

openoms, 2022-06-22:
With the ongoing DDOS on Tor (see status.torproject.org) having the latest package is important.

On RC2 the `tor --version` is the old one form Debian: https://packages.debian.org/bullseye/tor

The cause is that the Tor repo signing key has been not imported during the build. 
Should exit if unsuccesfull, but Tor might not be ready by then https://github.com/rootzoll/raspiblitz/blob/8f756e2d47f9bd3d1cc4318b5dda78872128b486/home.admin/config.scripts/tor.install.sh#L144

Will need to keep an eye on this at the next SDcard build. Might need to pull they key over clearnet.

in the meantime update Tor by pasting these to the CLI:

```
# variables
tor_deb_repo_clean="${tor_deb_repo#*tor+}"
tor_deb_repo_pgp_fingerprint="A3C4F0F979CAA22CDBA8F512EE8CBC9E886DDD89"

distribution=$(lsb_release -sc)
architecture=$(dpkg --print-architecture)

# Adding deb.torproject.org keyring
curl -s -x socks5h://127.0.0.1:9050 --connect-timeout 10 "${tor_deb_repo_clean}/torproject.org/${tor_deb_repo_pgp_fingerprint}.asc" | gpg --dearmor | sudo tee /etc/apt/trusted.gpg.d/torproject.gpg 

# Adding Tor Sources 
echo "
deb [arch=${architecture}] ${tor_deb_repo}/torproject.org ${distribution} main
deb-src [arch=${architecture}] ${tor_deb_repo}/torproject.org  ${distribution} main
" | sudo tee /etc/apt/sources.list.d/tor.list

sudo apt update

sudo apt install tor
```

The UPDATE menu is set to update from apt rather than source by default, but adding some improvement there to make sure the signing key is present.

gkucmierz, 2022-07-09:
```
Err:1 tor+http://apow7mjfryruh65chtdydfmqfpj5btws7nbocgtaovhvezgccyjazpqd.onion/torproject.org bullseye/main arm64 tor arm64 0.4.7.7-1~d11.bullseye+1
  Timed out while waiting to read 'first part of response' from proxy socks5h://127.0.0.1:9050 [IP: 127.0.0.1 9050]
E: Failed to fetch tor+http://apow7mjfryruh65chtdydfmqfpj5btws7nbocgtaovhvezgccyjazpqd.onion/torproject.org/pool/main/t/tor/tor_0.4.7.7-1%7ed11.bullseye%2b1_arm64.deb  Timed out while waiting to read 'first part of response' from proxy socks5h://127.0.0.1:9050 [IP: 127.0.0.1 9050]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
```

I have this error when I am trying to upgrade it.

PolishedViolet, 2022-07-09:
Just did a fresh install with the RC3 image.

Still an issue.

Just ran `./config.scripts/tor.install.sh install` again and now its up date `0.4.7.8` and RaspiBlitz is syncing fine.

rootzoll, 2022-07-12:
@openoms is there something to fix with the RC3 base image?

openoms, 2022-07-12:
need to check:
- [ ] Tor update option in the menu
- [ ] sudo apt update && sudo apt upgrade -y

openoms, 2022-07-12:
still in RC3:
```
₿ tor --version
Tor version 0.4.5.10.
```
as to change what we are shipping with would need the SDcard be rebuilt - which was not done since v1.8.0rc2, right?

So will need to either build the new SDcard image or make sure that Tor is updated to the latest version through apt on the first setup (might be preferable to do this anyway).

I'd like this to disappear too:
```
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: tor+http://apow7mjfryruh65chtdydfmqfpj5btws7nbocgtaovhvezgccyjazpqd.onion/torproject.org bullseye InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 74A941BA219EC810
W: Failed to fetch tor+http://apow7mjfryruh65chtdydfmqfpj5btws7nbocgtaovhvezgccyjazpqd.onion/torproject.org/dists/bullseye/InRelease  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 74A941BA219EC810
W: Some index files failed to download. They have been ignored, or old ones used instead.
```

rootzoll, 2022-07-12:
@openoms There was a rebuild of the sd card for v1.8.0rc3 .. but it seems it skipped the Tor update. How can we make sure that the updarte works on the sd card build for v1.8.0rc4?

rootzoll, 2022-07-13:
Just retested sd card build ... it just ends up with `Tor 0.4.5.10` This is a blocker for RC4 & final 1.8.0 build.

@openoms can you take a deeper look why the update to the new version is not happening during sd card build and give me a sign if fixed.

ktecho, 2022-07-13:
In x86, while using build_sdcard.sh in Ubuntu, I ended up with Tor version 0.4.7.8, but I guess it's different for ARM...

openoms, 2022-07-15:
On buster the above didn't help and I needed to add this key:
```
curl https://deb.torproject.org/torproject.org/A3C4F0F979CAA22CDBA8F512EE8CBC9E886DDD89.asc | gpg --import
gpg --export A3C4F0F979CAA22CDBA8F512EE8CBC9E886DDD89 | sudo apt-key add -
```

openoms, 2022-07-21:
OK solved in
v1.8.0rc4 :
```
₿ tor --version
Tor version 0.4.7.8.
Tor is running on Linux with Libevent 2.1.12-stable, OpenSSL 1.1.1n, Zlib 1.2.11, Liblzma 5.2.5, Libzstd 1.4.8 and Glibc 2.31 as libc.
Tor compiled with GCC version 10.2.1
```

closed 2022-07-21 by openoms