raspiblitz/raspiblitz #3181: LND v0.15.0 update

openoms, 2022-06-22:
v0.15.0-beta.rc6 was tagged 9 days ago. 
https://github.com/lightningnetwork/lnd/releases/

Tested and running ok on both aarch64 and amd64 so far.

openoms, 2022-06-24:
The stable is out: https://github.com/lightningnetwork/lnd/releases/tag/v0.15.0-beta

ktecho, 2022-06-24:
@openoms Have you tested if an upgrade in Raspiblitz 1.7.2 to this v0.15.0 is succesful?

openoms, 2022-06-24:
yes, as above was running the rc6 for a while

openoms, 2022-06-24:
@rootzoll just realized that when updating LND from the menu a full reboot is called. Is there any particular reason why this is necessary?

drmartinberger, 2022-06-28:
After updating Raspiblitz 1.7.2 to LND v0.15.0 the Raspiblitz menu entry `LND -> lnbalance` shows an error
`*** YOUR SATOSHI BALANCES ***`
`[lncli] flag provided but not defined: -1`
`parse error: Invalid numeric literal at line 1, column 10`
and the `LN EARNED (FORWARD) FEES` shows 0 sats incorrectly (probably a consequential error)

Maybe this is something to have an eye on or even a separate ticket?

openoms, 2022-06-28:
yes please open a separate issue as it is likely a problem with the script, not LND itself.

rootzoll, 2022-06-28:
@openoms so we just need to compile fresh protobuffs and thsi issue can go into final testing, correct?

drmartinberger, 2022-06-28:
> After updating Raspiblitz 1.7.2 to LND v0.15.0 the Raspiblitz menu entry `LND -> lnbalance` shows an error `*** YOUR SATOSHI BALANCES ***` `[lncli] flag provided but not defined: -1` `parse error: [...]

> yes please open a separate issue as it is likely a problem with the script, not LND itself.

Opened #3196 .

openoms, 2022-06-28:
> It just my habbit that when infrastructure setup is changed - a fresh reboot is a good routine. I know its some more minutes of downtime of the node, but I think it makes sense ... especially because we plan to have in the future more error analysis during startup - to detect & assist problems - so this would get utilized for such updates.

Ok, makes sense. I think it just bothers me on the machine I use to run a lot of custom services, not really applicable for a typically user or if it is they can work around using the scripts only as I do.

rootzoll, 2022-06-28:
OK I updated the protos - seem much smaller as before. I put this on  final testing & will make a test sd card build to run first tests

22388o, 2022-07-12:
Hey guys, I did upgrade to new version in my Raspiblitz and was no problem. Had reboot okay too.

rootzoll, 2022-07-27:
OK tested fine so far - will be part of v1.8.0 release. Closing issue.

closed 2022-07-27 by rootzoll